espressif/esp-idf #17420: OTA over HTTPS using Certificate Bundle slows down the file transfer

michaelsimp, 2025-08-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have OTA running in my application over https on an ESP32-S3 specifying the Root certificate as follows:
```
    esp_http_client_config_t config = {
        .url = CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL,
        .cert_pem = (char *)server_cert_pem_start,
        .timeout_ms = 5000, // CONFIG_EXAMPLE_OTA_RECV_TIMEOUT,
        .keep_alive_enable = true,
    };
```

I read I can use the certificate bundle feature which is my preference, so I enabled this in menu config "Certificate Bundle" using the "Enable trusted root certificate bundle" and "Use only the most common certificates from the default bundle" and modified my config to:
```
    esp_http_client_config_t config = {
        .url = CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL,
        //.cert_pem = (char *)server_cert_pem_start,
        .timeout_ms = 5000, // CONFIG_EXAMPLE_OTA_RECV_TIMEOUT,
        .crt_bundle_attach = esp_crt_bundle_attach, // Function to attach the certificate bundle other configurations
        .keep_alive_enable = true,
    };
```
This works but it is very slow doing OTA transfer.

The logs show repeated "esp-x509-crt-bundle: Certificate validated" which I suspect is the problem.  Is there any way to avoid this?
```
W (13:32:13.212) aOTAHttps: OTA HTTPS Task start
I (13:32:13.213) aOTAHttps: Running Advanced OTA Task. URI: https://xxxxxxxx.com/build/win.bin
I (13:32:13.215) aOTAHttps: OTA Attempt: 1 Free Stack NOT used (bytes): 3808
I (13:32:13.217) aOTAHttps: OTA started
W (13:32:13.220) aOTAHttps: OTA HTTPS Task running. Free Stack NOT used (bytes): 3712
I (13:32:16.367) esp-x509-crt-bundle: Certificate validated
I (26110) wifi:<ba-add>idx:0 (ifx:0, 00:17:13:20:bd:74), tid:1, ssn:2721, winSize:64
I (26660) wifi:<ba-add>idx:1 (ifx:0, 00:17:13:20:bd:74), tid:0, ssn:865, winSize:64
I (13:32:20.206) esp-x509-crt-bundle: Certificate validated
I (13:32:22.311) aOTAHttps: Connected to server
I (13:32:22.314) aOTAHttps: OTA image size: 2663872
I (13:32:22.315) aOTAHttps: Reading Image Description
I (13:32:22.317) aOTAHttps: Running software: Name: win  Ver: 2.1.3  Date: Aug 12 2025  Time: 13:15:55
W (13:32:22.319) aOTAHttps:     New software: Name: win  Ver: 2.1.1  Date: Jul 29 2025  Time: 14:36:49
I (13:32:22.321) aOTAHttps: Verifying chip id of new image: 9
I (13:32:22.377) aOTAHttps: Flash write: 1024 bytes
I (13:32:25.131) esp-x509-crt-bundle: Certificate validated
13:32:26 [HOST] WIN status = 1: Config Error
I (13:32:31.726) esp-x509-crt-bundle: Certificate validated
I (13:32:37.432) esp-x509-crt-bundle: Certificate validated
I (13:32:39.762) aOTAHttps: Flash write: 52224 bytes
I (13:32:43.253) esp-x509-crt-bundle: Certificate validated
13:32:47 [HOST] WIN status = 1: Config Error
I (13:32:49.148) esp-x509-crt-bundle: Certificate validated
I (13:32:53.806) esp-x509-crt-bundle: Certificate validated
I (13:32:56.147) aOTAHttps: Flash write: 103424 bytes
...
```
If I specify a root certificate instead of using esp_crt_bundle_attach, it runs much faster:
```
W (13:37:11.365) aOTAHttps: OTA HTTPS Task start
I (13:37:11.367) aOTAHttps: Running Advanced OTA Task. URI: https://192.168.1.2:8070/build/win.bin
I (13:37:11.370) aOTAHttps: OTA Attempt: 1 Free Stack NOT used (bytes): 3768
I (13:37:11.372) aOTAHttps: OTA started
W (13:37:11.381) aOTAHttps: OTA HTTPS Task running. Free Stack NOT used (bytes): 3320
I (35056) wifi:<ba-add>idx:0 (ifx:0, 00:17:13:20:bd:74), tid:0, ssn:1533, winSize:64
I (13:37:12.282) aOTAHttps: Connected to server
I (13:37:12.283) aOTAHttps: OTA image size: 2678976
I (13:37:12.285) aOTAHttps: Reading Image Description
I (13:37:12.897) aOTAHttps: Running software: Name: win  Ver: 2.1.3  Date: Aug 12 2025  Time: 13:15:55
W (13:37:12.899) aOTAHttps:     New software: Name: win  Ver: 2.1.3  Date: Aug 12 2025  Time: 12:18:48
I (13:37:12.901) aOTAHttps: Verifying chip id of new image: 9
I (13:37:12.957) aOTAHttps: Flash write: 1024 bytes
I (13:37:15.231) aOTAHttps: Flash write: 52224 bytes
I (13:37:18.355) aOTAHttps: Flash write: 103424 bytes
I (13:37:20.821) aOTAHttps: Flash write: 154624 bytes
I (13:37:22.895) aOTAHttps: Flash write: 205824 bytes
I (13:37:25.681) aOTAHttps: Flash write: 257024 bytes
I (13:37:27.656) aOTAHttps: Flash write: 308224 bytes
...
```

michaelsimp, 2025-08-12:
Ok I was mistaken, too many changes at once.  
I was running OTA from a local HTTS server on my LAN, and now I need to run this from a Linux HTTPS server on the WAN,
I switched back to using the Root certificate instead of Certificate Bundle and it is still working very slowly.
My bin file is only 2.6MB and downloads using my browser immediatly, its just my OTA which is slow.
I suspect the LAN based HTTPS server OTA is also slower than it should be but its faster than the WAN.

Should the library log the following regularly through the OTA download when using certificate bundle?
`I (13:32:31.726) esp-x509-crt-bundle: Certificate validated
`

I set the buffer_size as follows:

```
    esp_http_client_config_t config = {
        .url = CONFIG_EXAMPLE_FIRMWARE_UPGRADE_URL,
        .cert_pem = (char *)server_cert_pem_start,
        .timeout_ms = 5000, // CONFIG_EXAMPLE_OTA_RECV_TIMEOUT,
        .buffer_size = 32768,
        // .crt_bundle_attach = esp_crt_bundle_attach, // Function to attach the certificate bundle other configurations
        .keep_alive_enable = true,
    };

```
...and the buffer_caps as follows:
```
    esp_https_ota_config_t ota_config = {
        .http_config = &config,
        .http_client_init_cb = _http_client_init_cb, // Register a callback to be invoked after esp_http_client is initialized
#ifdef CONFIG_EXAMPLE_ENABLE_PARTIAL_HTTP_DOWNLOAD
        .partial_http_download = true,
        .max_http_request_size = CONFIG_EXAMPLE_HTTP_REQUEST_SIZE,
        .buffer_caps = MALLOC_CAP_INTERNAL,
#endif
```
By the way I have CONFIG_EXAMPLE_ENABLE_PARTIAL_HTTP_DOWNLOAD defined true, and the CONFIG_EXAMPLE_HTTP_REQUEST_SIZE set to 16384.  I tried setting it higher in menu config, but it keeps reverting back to 16384.

Can you suggest and other any other diagnostics or settings to try? Here is a log showing how long it takes. Roughly how long would you expect it to take?

```
W (19:17:58.355) aOTAHttps: OTA HTTPS Task start
I (19:17:58.356) aOTAHttps: Running Advanced OTA Task. URI: https://mqtt.fuel-trans.com/FuelTransSM/win/win.bin
I (19:17:58.358) aOTAHttps: OTA Attempt: 0 Free Stack NOT used (bytes): 3756
I (19:17:58.360) aOTAHttps: OTA started
W (19:17:58.368) aOTAHttps: OTA HTTPS Task running. Free Stack NOT used (bytes): 3708
I (27268) wifi:<ba-add>idx:1 (ifx:0, 00:17:13:20:bd:74), tid:0, ssn:1606, winSize:64
I (19:18:09.015) aOTAHttps: Connected to server
I (19:18:09.015) aOTAHttps: OTA image size: 2663872
I (19:18:09.017) aOTAHttps: Reading Image Description
I (19:18:09.019) aOTAHttps: Running software: Name: win  Ver: 2.1.3  Date: Aug 12 2025  Time: 13:57:30
W (19:18:09.020) aOTAHttps:     New software: Name: win  Ver: 2.1.1  Date: Jul 29 2025  Time: 14:36:49
I (19:18:09.022) aOTAHttps: Verifying chip id of new image: 9
I (19:18:09.071) aOTAHttps: Flash write: 1024 bytes
I (19:18:22.640) aOTAHttps: Flash write: 52224 bytes
I (19:18:36.151) aOTAHttps: Flash write: 103424 bytes
I (19:18:49.620) aOTAHttps: Flash write: 154624 bytes
I (19:19:03.361) aOTAHttps: Flash write: 205824 bytes
I (19:19:17.340) aOTAHttps: Flash write: 257024 bytes
I (19:19:30.681) aOTAHttps: Flash write: 308224 bytes
I (19:19:44.742) aOTAHttps: Flash write: 359424 bytes
I (19:20:06.731) aOTAHttps: Flash write: 410624 bytes
I (19:20:22.292) aOTAHttps: Flash write: 461824 bytes
I (19:20:36.751) aOTAHttps: Flash write: 513024 bytes
I (19:20:53.862) aOTAHttps: Flash write: 564224 bytes
I (19:21:08.611) aOTAHttps: Flash write: 615424 bytes
I (19:21:22.600) aOTAHttps: Flash write: 666624 bytes
I (19:21:38.342) aOTAHttps: Flash write: 717824 bytes
I (19:21:51.790) aOTAHttps: Flash write: 769024 bytes
I (19:22:09.601) aOTAHttps: Flash write: 820224 bytes
I (19:22:24.620) aOTAHttps: Flash write: 871424 bytes
I (19:22:38.861) aOTAHttps: Flash write: 922624 bytes
[... 25 lines cut ...]
I (19:29:21.411) aOTAHttps: Flash write: 2253824 bytes
I (19:29:34.770) aOTAHttps: Flash write: 2305024 bytes
I (19:29:52.351) aOTAHttps: Flash write: 2356224 bytes
I (19:30:07.233) aOTAHttps: Flash write: 2407424 bytes
I (19:30:25.011) aOTAHttps: Flash write: 2458624 bytes
I (19:30:40.891) aOTAHttps: Flash write: 2509824 bytes
I (19:30:54.122) aOTAHttps: Flash write: 2561024 bytes
I (19:31:08.730) aOTAHttps: Flash write: 2612224 bytes
I (19:31:29.541) aOTAHttps: Flash write: 2663424 bytes
I (19:31:29.546) esp_image: segment 0: paddr=00410020 vaddr=3c1d0020 size=9ac14h (633876) map
I (19:31:29.637) esp_image: segment 1: paddr=004aac3c vaddr=3fca1f00 size=053dch ( 21468) 
I (19:31:29.643) esp_image: segment 2: paddr=004b0020 vaddr=42000020 size=1cbf5ch (1883996) map
I (19:31:29.910) esp_image: segment 3: paddr=0067bf84 vaddr=3fca72dc size=00784h (  1924) 
I (19:31:29.913) esp_image: segment 4: paddr=0067c710 vaddr=40374000 size=1de58h (122456) 
I (19:31:29.935) esp_image: segment 5: paddr=0069a570 vaddr=600fe000 size=00020h (    32) 
I (19:31:29.943) esp_image: segment 0: paddr=00410020 vaddr=3c1d0020 size=9ac14h (633876) map
I (19:31:30.033) esp_image: segment 1: paddr=004aac3c vaddr=3fca1f00 size=053dch ( 21468) 
I (19:31:30.039) esp_image: segment 2: paddr=004b0020 vaddr=42000020 size=1cbf5ch (1883996) map
I (19:31:30.314) esp_image: segment 3: paddr=0067bf84 vaddr=3fca72dc size=00784h (  1924) 
I (19:31:30.318) esp_image: segment 4: paddr=0067c710 vaddr=40374000 size=1de58h (122456) 
I (19:31:30.337) esp_image: segment 5: paddr=0069a570 vaddr=600fe000 size=00020h (    32) 
I (19:31:30.378) aOTAHttps: Boot partition updated. Next Partition: 16
I (19:31:30.382) aOTAHttps: OTA finish
W (19:31:30.383) aOTAHttps: ESP_HTTPS_OTA upgrade successful. Rebooting ...
```

mahavirj, 2025-08-12:
@michaelsimp 

> By the way I have CONFIG_EXAMPLE_ENABLE_PARTIAL_HTTP_DOWNLOAD defined true, and the CONFIG_EXAMPLE_HTTP_REQUEST_SIZE set to 16384.

Please try disabling partial HTTP download feature. With partial HTTP download feature, for every new range request a TLS connection is opened. This is likely to slow down the OTA update. We are investigating on the ways to optimize this behavior.

KaeLL, 2025-08-12:
@michaelsimp I'm afraid you'll have to dig deeper to make sure the buffers from lower layers (esp-http-client, MBedTLS, LwIP, PHY) aren't choking the data transfers. I hit this problem a couple years back and decided not to pursuit it because I deemed it too costly to do so.