espressif/esp-idf #17428: Installation fails because it thinks I'm using Python 3.8.7, when I have a fresh install of 3.13.6 installed

SoLongSidekick, 2025-08-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-tools-setup-2.3

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Uh for it to actually install?

### What is the actual behavior?

Install fails and I can't copy what it's telling me but it says something along the lines of "installation failed due to reason 1, ESP supports Python version 3.9 and you have version 3.8.7 installed". I literally just wiped everything and installed a fresh Python install with the newest version.

### Steps to reproduce.

1. Try to install ESP-IDF
2. Fail.

### Build or installation Logs.

```plain
2025-08-12 20:45:39.965   Log opened. (Time zone: UTC-07:00)
2025-08-12 20:45:39.965   Setup version: Inno Setup version 6.2.0
2025-08-12 20:45:39.965   Original Setup EXE: C:\Users\devte\Downloads\esp-idf-tools-setup-2.3.exe
2025-08-12 20:45:39.965   Setup command line: /SL5="$F0546,2435469,831488,C:\Users\devte\Downloads\esp-idf-tools-setup-2.3.exe" 
2025-08-12 20:45:39.965   Windows version: 10.0.19045  (NT platform: Yes)
2025-08-12 20:45:39.965   64-bit Windows: Yes
2025-08-12 20:45:39.965   Processor architecture: x64
2025-08-12 20:45:39.965   User privileges: None
2025-08-12 20:45:39.975   Administrative install mode: No
2025-08-12 20:45:39.975   Install mode root key: HKEY_CURRENT_USER
2025-08-12 20:45:39.975   64-bit install mode: Yes
2025-08-12 20:45:39.985   Created temporary directory: C:\Users\devte\AppData\Local\Temp\is-CIDF8.tmp
2025-08-12 20:45:40.025   -- DLL function import --
2025-08-12 20:45:40.025   Function name: idpAddFile
2025-08-12 20:45:40.025   DLL name: files:idp.dll
2025-08-12 20:45:40.025   Extracting temporary file: C:\Users\devte\AppData\Local\Temp\is-CIDF8.tmp\idp.dll
2025-08-12 20:45:40.035   Dest DLL name: C:\Users\devte\AppData\Local\Temp\is-CIDF8.tmp\idp.dll
2025-08-12 20:45:40.035   Importing the DLL function.
2025-08-12 20:45:40.155   Successfully imported the DLL function. Delay loaded? No
2025-08-12 20:45:40.155   -- DLL function import --
2025-08-12 20:45:40.155   Function name: idpAddFileComp
2025-08-12 20:45:40.155   DLL name: files:idp.dll
2025-08-12 20:45:40.155   Dest DLL name: C:\Users\devte\AppData\Local\Temp\is-CIDF8.tmp\idp.dll
2025-08-12 20:45:40.155   Importing the DLL function.
2025-08-12 20:45:40.155   Successfully imported the DLL function. Delay loaded? No
2025-08-12 20:45:40.155   -- DLL function import --
2025-08-12 20:45:40.155   Function name: idpAddMirror
2025-08-12 20:45:40.155   DLL name: files:idp.dll
2025-08-12 20:45:40.155   Dest DLL name: C:\Users\devte\AppData\Local\Temp\is-CIDF8.tmp\idp.dll
2025-08-12 20:45:40.155   Importing the DLL function.
2025-08-12 20:45:40.155   Successfully imported the DLL function. Delay loaded? No
2025-08-12 20:45:40.155   -- DLL function import --
[... 569 lines cut ...]
2025-08-12 20:59:31.610   idf_tools.py command: C:\.espressif\tools\idf-python\3.8.7\python.exe "C:\ESP-IDF\\tools\idf_tools.py" --idf-path C:\ESP-IDF\
2025-08-12 20:59:31.610   Installing tools:C:\.espressif\tools\idf-python\3.8.7\python.exe "C:\ESP-IDF\\tools\idf_tools.py" --idf-path C:\ESP-IDF\ install
2025-08-12 20:59:32.970   Done, exit code=1
2025-08-12 20:59:32.970   --------
2025-08-12 20:59:32.970   Running command: C:\.espressif\tools\idf-python\3.8.7\python.exe "C:\ESP-IDF\\tools\idf_tools.py" --idf-path C:\ESP-IDF\ install
                          ESP-IDF supports Python 3.9 or newer but you are using Python 3.8.7. Please upgrade your installation as described in the documentation.
                          
2025-08-12 20:59:32.970   --------
2025-08-12 20:59:32.970   Message box (OK):
                          Installation has failed with exit code 1
2025-08-12 21:03:42.761   User chose OK.
2025-08-12 21:03:42.791   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2025-08-12 21:03:45.581   User chose Yes.
2025-08-12 21:03:45.871   CurStepChanged raised an exception.
```

### Diagnostic report archive.

Not possible because the install fails

### More Information.

Nope

SoLongSidekick, 2025-08-14:
OK, here's where I currently sit and what I still need help with (questions highlighted in bold): I completely wiped all ESP stuff from my PC and installed the worldwide v5.5 you linked above. It seems to have installed fine except for the very last part which leaves me with the following in the ESP-IDF command window:

`Using Python in C:\Espressif\python_env\idf5.5_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.44.0\cmd\
git version 2.44.0.windows.1
Activating ESP-IDF 5.5
Setting IDF_PATH to 'C:\ESP-IDF'.
* Checking python version ... 3.11.2
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... cmd.exe
* Detecting outdated tools in system ... Found tools that are not used by active ESP-IDF version.
For removing old versions of amazon-corretto-11-x64-windows-jdk, espressif-ide use command 'python.exe C:\ESP-IDF\tools\idf_tools.py uninstall'
To free up even more space, remove installation packages of those tools.
Use option python.exe C:\ESP-IDF\tools\idf_tools.py uninstall --remove-archives.
'C:\Users\QECHIC~1\AppData\Local\Temp\esp_idf_activate_QE' is not recognized as an internal or external command,
operable program or batch file.

C:\ESP-IDF>`

**Wouldn't running the command "python.exe C:\ESP-IDF\tools\idf_tools.py uninstall" like it's telling me I should just uninstall what I just finished installing?** Or am I misunderstanding something?

**I'm still confused about the Visual Studio Code ESP-IDF extension, which I'd really like to also have installed because I often use Platform.IO. Is there a way to have the extension share the same files as the normal ESP-IDF installation?** I had to install ~6 vitally important programs in order to come close to installing it on top of the normal ESP-IDF install earlier, thinking once it installed fully I would get at least the majority of that hard drive space back due to installation files being deleted or similar. But to my horror, nope. Something like 17GB of space GONE and the install still couldn't make it all the way to the end due to running out of hard drive space.

jakub-kocka, 2025-08-14:
Can imagine the headache, sorry for that @SoLongSidekick ... unfortunately, this is a bit out of scope, as it is a real issue with Installer/ESP-IDF/Extension
It is more like a customer support issue, but let's finish what we have started :)

It is completely fine to run `python.exe C:\ESP-IDF\tools\idf_tools.py uninstall` or `C:\Espressif\python_env\idf5.5_py3.11_env\Scripts\python.exe C:\ESP-IDF\tools\idf_tools.py uninstall` when ESP-IDF exported (/ opened the CMD/PowerShell ESP-IDF)
... during the process of exporting, it updates some tools or from the last installs something stayed, it will just remove the old versions as it is telling you **For removing old versions** you can even additionally use the parameter `--remove-archives` -

> To free up even more space, remove installation packages of those tools.

Secondly, the ESP-IDF Extension for VS Code is another thing.

It is important to understand that you are not installing an application when installing the ESP-IDF, you are installing the Development Framework.
What Installer does is that it downloads for you all the toolchains, toolsets, etc., and prepares an environment for you so you can use it for your development.

So when you follow the official guide, you should be able to get it working altogether. https://docs.espressif.com/projects/vscode-esp-idf-extension/en/latest/installation.html

You can choose to use the already present framework (ESP-IDF) or you can install a new one ... the same you can do with Installer, you can run it again and install another framework and another, or choose the existing one and just install the missing necessary things.

Hope this clears things up for you.