espressif/esp-idf #17430: ESP-IDF CLI on Windows: Broken encoding, fragile scripts, and poor automation support

Masquelier, 2025-08-13:
**Title:** ESP-IDF CLI on Windows: Broken encoding, fragile scripts, and poor automation support

**Environment:**
- OS: Windows 10 / 11 (native, no WSL)
- Shell: PowerShell 5.1 (default)
- ESP-IDF version: v5.5
- Python: 3.13.6 (used by ESP-IDF installer)
- Installation path: C:\Espressif\frameworks\esp-idf-v5.5
- Project path: C:\Users\pierr\Documents\ESP_PM\adc_project

**Summary:**
ESP-IDF CLI is the only interface that exposes the full feature set of the framework, yet its integration on Windows is unstable and poorly supported. PowerShell scripts are fragile, encoding is broken in standard consoles, and `.bat` automation is unreliable.

**Issues:**
1. `export.ps1` outputs corrupted characters in PowerShell 5.1, even with `chcp 65001` and `[Console]::OutputEncoding = UTF8`.
2. `.bat` scripts calling PowerShell with `-NoExit` still close immediately or fail to display readable output.
3. ESP-IDF messages (e.g. “Build réussi”) appear as `Ô£à Build r├®ussi.` in the console.
4. `idf.py` fails silently or with misleading errors when called from `.bat` or embedded PowerShell.
5. ESP-IDF CLI assumes Linux-like behavior and does not adapt to Windows shell limitations.
6. VS Code and Eclipse integrations are limited and do not expose full CLI functionality.

**Steps to Reproduce:**
1. Install ESP-IDF v5.5 using `install.ps1`
2. Create a minimal project with valid `CMakeLists.txt`
3. Write a `.bat` file that calls PowerShell to activate ESP-IDF and run `idf.py build`
4. Observe corrupted output and inconsistent behavior

**Example `.bat` script:**
```bat
@echo off
chcp 65001 >nul
powershell.exe -NoExit -ExecutionPolicy Bypass -Command "[Console]::OutputEncoding = [System.Text.Encoding]::UTF8; $env:IDF_PATH='C:\Espressif\frameworks\esp-idf-v5.5'; . \"$env:IDF_PATH\export.ps1\"; Set-Location 'C:\Users\pierr\Documents\ESP_PM\adc_project'; Write-Host 'Compilation ESP-IDF...'; idf.py build; pause"

jakub-kocka, 2025-08-15:
Hello, @Masquelier,

I noticed that your current approach calls PowerShell from within a Batch file. This may introduce unnecessary complexity or potential compatibility issues (Maybe Batch has some issues since MS-DOS).
Since you are already using PowerShell during the installation process, you might consider using PowerShell scripting directly instead.

Additionally, the functionality you are aiming for appears to be available in the ESP-IDF source code (framework directory) through the `export` scripts, which activate the ESP-IDF environment.

You could use these as a base and adapt them for your requirements:

- Batch version - https://github.com/espressif/esp-idf/blob/master/export.bat
- PowerShell version - https://github.com/espressif/esp-idf/blob/master/export.ps1

It won't be necessary to change the encoding then, but if you still wish to try changing the encoding, here’s a reference source which might point to something (using the same approach): https://www.elevenforum.com/t/batch-file-character-encoding-problem.24616/
```batch
@ECHO OFF
REM Set the desired code page (e.g., UTF-8)
chcp 65001 > nul

Run your .bat file
CALL yoursbatchfile.bat
```

closed 2025-08-22 by dobairoland

jakub-kocka, 2025-08-25:
Adding another related link (https://exploratory.io/note/exploratory/Enabling-UTF-8-on-Windows-hYc3yWL0) - it could be produced because UTF-8 encoding might not be set on Windows.

And closing this for inactivity.