espressif/esp-idf #17431: time switch

diplfranzhoepfinger, 2025-08-13:
### Is your feature request related to a problem?

Hi, 

is there a "time switch" Function ? 

like i want the ESP to fire me every Day at 12:05

diplfranzhoepfinger, 2025-08-13:
scheduler seems to be a common other name for time switch

ESP-Marius, 2025-08-14:
You mean you want a timer that is periodically called, but based on wall time instead of ms/us, right?

Since this something that can be achieved with the tools we have (e.g. by using esp_timer and `time.h` functions to convert from target wall-time to ms) and is probably not a very common use-case for the timers, Im not sure it makes sense to integrate into IDF itself.