espressif/esp-idf #17432: ESP32-C6/H2 USB_SERIAL_JTAG not initialised correctly when logging is disabled

Skyler84, 2025-08-13:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v6.0-dev-1489-g4e036983a7

### Espressif SoC revision.

ESP32-C6/H2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC1

### Power Supply used.

USB

### What is the expected behavior?

When using USB_SERIAL_JTAG as the primary console, all application output/input should be directed over that USB connection. If the device is reset using the RST button, by Hangup over UART or by Hangup over USB_SERIAL_JTAG CDC, the device should initialise the USB_SERIAL_JTAG CDC and output any data over that. If the device is plugged in to that port, it should work as soon as it has been enumerated by the host and is connected using a serial terminal application.

### What is the actual behavior?

When starting `idf.py monitor` the device is reset as expected and output over the USB_SERIAL_JTAG works. When the device is reset in hardware, the USB_SERIAL_JTAG phy is correctly put into reset but appears to not be properly initialised IF LOGGING IS DISABLED. if logging is enabled, it works just fine, however it should work  

### Steps to reproduce.

1. Build the `usb_serial_echo` example, disable logging in menuconfig
2. flash device
3. open `idf.py monitor` and observe correct interaction over serial terminal.
4. reset device using  HARDWARE RESET (not Ctrl+T,Ctrl+R) or by unplugging and replugging the USB (not UART) port, ensuring device is fully powered down.
5. observe that nothing happens over the USB_SERIAL_JTAG interface and no output/input happens. (incorrect behaviour)
6. reset device using Ctrl+T,Ctrl+R
7. observe that serial monitor now behaves correctly again.
...

### Debug Logs.

```plain
N/A, only reproducable without logs as far as i can tell.
```

### Diagnostic report archive.

### More Information.

It looks like the console is trying to fallback to the secondary output because the USB_SERIAL_JTAG is not enumerated by the host in time for the application?

Alvin1Zhang, 2025-09-09:
@Skyler84 Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

Skyler84, 2025-09-09:
I will do some more testing, but this is probably a good source of info for now. I was originally using usb-serial-jtag for ot_rcp and that's where the issue originally arose