espressif/esp-idf #17438: component should be excluded if the containing component was not required/included

wuyuanyi135, 2025-08-14:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am maintaining a megarepo that contains lots of shared components and firmwares for various of different products (driven by ESP chips). The project layout is like this:
```
megarepo 
|
| -- components
        | -- my_usb
        | -- my_wifi
        | ...
| -- projects
        | -- project_device_0
        | -- project_device_1
        | -- project_device_2
        | ...
```

I have noticed that for the `project_device_x`, even if I have `idf_build_set_property(MINIMAL_BUILD ON)`, those components not in `REQUIRES` entry of the `CMakelists.txt` of the `main` component will still compile, and their `idf_component.yml` will still download. This cause waste of compilation time AND severe issues when the component was not intended for some project.

For example, `my_usb` depends on `esp_tinyusb` which was not supported by esp32c3 but one of my board is esp32c3. CMake reload failed because it tells me that the hosted component `esp_tinyusb` does not support ESP32C3. To solve this, I have to make use of `rules` in `idf_component.yml` to make sure it won't be downloaded and checked.

Another example, suppose I have a huge codebase that has to be compiled for long long time and it was not even used by some projects. I found that regardless of listing in the REQUIRES entry or not, the compile won't be skipped. 

The only working solution I found now is to use set(EXCLUDE_COMPONENTS my_usb ...) to explicitly remove the components from entering my component list. This means if later new components were added to my megarepo I have to update every root `CMakeLists.txt` that should ignore the new component. This feels very weird and cumbersome.

I suppose `MINIMAL_BUILD` should have done the work including only the `REQUIRES` components in the build list. Why wouldn't this work? Am I missing something?

Thanks

wuyuanyi135, 2025-08-14:
@igrr Thanks for the prompt reply. Thanks for confirming the usage of `EXCLUDE_COMPONENTS`. This was not optimal because of newly added component will require updating the existing CMakeLists.

As for the KConfig options, I did try to use the Kconfig option to make the external dependency ignored with `rules: if...` but it didn't work. It did not find my variable ($CONFIG{xxx} was set correctly). I will give it a try right now.

wuyuanyi135, 2025-08-14:
@igrr Thanks. I saw this default. If my project `main` does **NOT** depend on `A` component, and `A` component depends on `B` component, compiling the project still have `B` compiled. Is this expected behavior?

wuyuanyi135, 2025-08-14:
Sorry for bother you again but I also noticed that Kconfig variable was not working: I defined the Kconfig below
```
menu "UniLAB BT"
    config UNILAB_BT_ENABLED
    bool "Enable UniLAB Bluetooth Nimble library"
    default n
    select

endmenu
```

and let the component to be dependent on this option:

```
  h2zero/esp-nimble-cpp:
    version: "^2.3.1"
    rules:
      - if: "$CONFIG{UNILAB_BT_ENABLED}"
```

It seems there is a cyclic dependency: the idf_component.yml failed the CMake reload because `CONFIG_UNILAB_BT_ENABLED` was not yet generated in sdkconfig. Meanwhile because of failing CMake reload, CONFIG_UNILAB_BT_ENABLED was not able to be generated. I noticed that Kconfig variable was supported in 6.0. Mine was on release 5.5, probably this was the issue?

igrr, 2025-08-14:
That's odd, actually it should work in 5.5.0 as well: https://github.com/espressif/esp-idf/commit/b43b2001a28c3ccccc594f1cb840c47d5d7ff46f

Could you please try to narrow this down to a small example which reproduces the issue and which you can share?

igrr, 2025-08-15:
Actually after fixing the error, I don't get any error output:
- If the Kconfig option is disabled, the project configures correctly, no dependency is downloaded.
- After enabling the option menuconfig, and running idf.py reconfigure twice, the dependency gets downloaded.

The only remaining issue IMO is the fact that we need to run reconfigure (or build) twice for the change to the option to get picked up.

Just in case, here is the modified MRE project.

[MRE-IDFGH-16286.zip](https://github.com/user-attachments/files/21792824/MRE-IDFGH-16286.zip)

wuyuanyi135, 2025-08-15:
@igrr Thank you. I also pushed the fixed MRE. However, if I don't want to require this project at the first place (this small project won't use my_nimble). this will cause idf.py reconfigure to fail because the variable is not created. Is this intended behavior?

hfudev, 2025-08-26:
Yes, you're right. We are improving it internally.

As a workaround, the dependency components listed in `my_nimble/idf_component.yml` will only be downloaded and added to `REQUIRES` if the `if` conditions are met. Therefore, you could move all the conditions into `rules/matches` to control this behavior.