espressif/esp-idf #17440: MCPWM comparator update doesn't work unless RST caused by idf.py flash or idf.py monitor

CMakeMain, 2025-08-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.4

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom PCB

### Power Supply used.

USB

### What is the expected behavior?

I am using MCPWM and LAP to drive four motors. I have split my code to act like a command. My main task is supposed to recieve commands from my remote controller with my camera. However, when I attempt to update my MCPWM comparator from within my mutex it doesn't seem to update at all unless if I call ```idf.py monitor``` or ```idf.py flash``` even if I don't use them meaningfully. I am super confused as to why this happens as well as the fact that it always breaks when the move motor function is put into the lock.

### What is the actual behavior?

Because I want the motors to brake by default so it doesn't move before the software finished initialises, it should run at first at 50% dc, but later change. However, it seems to get stuck on the initial 50% dc value. I have confirmed the loop isn't getting stuck, but the values don't seem to be updating in the global speed buffer.

### Steps to reproduce.

1. Use this motor init code:
motors.h
```
#ifndef MOTORS_H
#define MOTORS_H

#include <stdint.h>
#include <stddef.h>
#include <math.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "driver/mcpwm_timer.h"
#include "driver/mcpwm_oper.h"
#include "driver/mcpwm_gen.h"
#include "driver/mcpwm_cmpr.h"
#include "driver/gpio.h"
#include "driver/gptimer.h"
#include "esp_log.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "utils.h"
#include "statedefs.h"
#include "comms_i2c.h"

// defines
[... 290 lines cut ...]
```
2. Initialise it in a simple task like the following, with a binary semaphore as a notification from the controller
``` 
static void task(void* pParams){
  motors_init();

  for(;;){
    if(xSemaphoreTake(blahblah, portMAX_DELAY)){
      // either move_vector() or move_polar
      motors_move();
    }
  }
}
```
4. Use a logic analyser and check the PWM waveforms. They should not change even if your input duty cycle is changed.

### Debug Logs.

```plain

```

### More Information.

As soon as I put it out side the mutex, it works but I didn't want to risk a race condition as there were other tasks runing sensor fusion on the same data struct the motors used.

CMakeMain, 2025-08-19:
I have tried simple ticks and it seems to work okay, but I am very confused as to why my code doesn't work. The inputs are sound and it only seems to work after calling idf.py monitor or idf.py flash without turning off the power. I am not sure how that could be my code issue as it seems to be a very specific issue?

Kainarx, 2025-08-19:
It is indeed a strange question. Ok, let's gradually narrow down the suspicious code, for instance, remove the code of bno055. 
And just calculate a trigonometric function randomly in the app main to replace hard coding value.

CMakeMain, 2025-08-19:
Hi,

I have another update. I have tested the following main function:

void app_main(void)
{
    //puts("Starting main task...\n");
    //xTaskCreatePinnedToCore(main_task, "Main Task", 16000, NULL, configMAX_PRIORITIES - 1, NULL, 0);
    
    motors_init();
    float yaw_offset = 0.0f;
    float yaw_raw = 0.0f;
    float true_yaw = 0.0f;
    bno_comms_init(BNO055_PORT);
    struct bno055_t config_settings = {0};
    init_bno055(&config_settings);
    // determine initial offset
    bno055_convert_float_euler_h_deg(&yaw_offset);
    MOTORS_ON
    float true_correction = 0.0f;
    //float test_yaw = -20.0f;
    int64_t time;
    for(;;){
        bno055_convert_float_euler_h_deg(&yaw_raw);
        true_yaw = fmodf(yaw_raw - yaw_offset + 360.0f, 360.0f);
        //true_correction = imu_correction(PID_DEFAULT, 0.0, true_yaw);
        time = esp_timer_get_time();
        true_correction = sinf((float) time);
        //ESP_LOGI("Main", "Correction amt: %f", true_correction);
        MOTORS_BRAKE(true_correction);
        //ESP_LOGI("MAIN", "True yaw: %f", true_yaw);
        //move_polar(0.0f, 1.0f, imu_correction(PID_DEFAULT, 0.0f, true_yaw));
        move_motors();
        //vTaskDelay(pdMS_TO_TICKS(200));
        //test_yaw *= -1.0f;
    }
}

And it works perfectly fine. I have also increased the main app stack size to 4000. I am really confused as to what the issue could be now. The motors move perfectly, however they are erratic given the rate at which they are changing. However, with the BNO055, it doesn't seem to move at all / update the duty cycle.

CMakeMain, 2025-08-20:
I have also tested the function with the random value generated by the trigonometric function and it works perfectly fine. However, when I substitute it back with the yaw data taken from the BNO055 it still fails to work.

Kainarx, 2025-08-20:
Maybe you need to check whether this variable has been unexpectedly optimized by the compiler after using BNO055

CMakeMain, 2025-08-20:
I have made the respesctive yaw variables volatile to no avail. Oddly enough, this same issue doesn't apply to the trigonometric function which doesn't make sense to me. I am just genuienly confused. I have also added a delay to no avail after the initialisation of the BNO055 to isolate issues regarding the gyro not being ready but it proves not to be the case. Could there be any other possible reason?

CMakeMain, 2025-08-21:
I must close this now. Turns out it was my fault. I dont know why but it fixed itself. I need to find out if there are any memory corruption issues

closed 2025-08-21 by CMakeMain