espressif/esp-idf #17458: The instructions for running the security_features_app example in Qemu are not working

friissoren, 2025-08-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I am trying to run the security_features_app example in Qemu:
https://github.com/espressif/esp-idf/tree/master/examples/security/security_features_app

The readme in general is very detailed and easy to follow but I cannot get the [last section](https://github.com/espressif/esp-idf/tree/master/examples/security/security_features_app#enable-security-features-with-help-of-qemu) about running the full example, including encrypted partitions etc., to work in Qemu.

### What is the actual behavior?

The provided command `idf.py merge-bin --merge-args ../qemu/qemu_flash_args -o qemu/security_features_flash_image.bin` does not work. It fails with the error `Error: No such option: --merge-args`.

### Steps to reproduce.

Diving into the source code of the idf.py, serial_ext.py and the esptool, it is pretty clear that the dashes in front of `merge-args` are wrong. Removing those (and also giving the -o parameter earlier, so that the idf.py translated command string to the esptool is not given twice) gives a new error:

```
$ idf.py merge-bin -o qemu/security_features_flash_image.bin merge-args ../qemu/qemu_flash_args
Adding "merge-bin"'s dependency "all" to list of commands with default set of options.
Executing action: all (aliases: build)
Running ninja in directory /home/<user>/SW/esp/test_projects/security_features_app/build
Executing "ninja all"...
[1/4] cd /home/<user>/SW/esp/test_projects/security_features_app/build/esp-idf/esptool_py && /home/<user>/.espressif/pytho...artition_table/partition-table.bin /home/<user>/SW/esp/test_projects/security_features_app/build/security_features_app.bin
security_features_app.bin binary size 0x41000 bytes. Smallest app partition is 0x100000 bytes. 0xbf000 bytes (75%) free.
[1/2] cd /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/esp-idf/esptool_py && /home/<user>/.espr....py --offset 0xd000 bootloader 0x0 /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/bootloader.bin
Bootloader binary size 0xb000 bytes. 0x2000 bytes (15%) free.
[2/2] cd /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/esp-idf/esptool_py && /home/<user>/.espr....py --offset 0xd000 bootloader 0x0 /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/bootloader.bin
Bootloader binary size 0xb000 bytes. 0x2000 bytes (15%) free.
[4/4] Completed 'bootloader'Executing action: merge-bin
Merged binary qemu/security_features_flash_image.bin will be created in the build directory...
Running merge_bin in directory /home/<user>/SW/esp/test_projects/security_features_app/build
Executing "/home/<user>/.espressif/python_env/idf5.5_py3.12_env/bin/python -m esptool --chip esp32s3 merge_bin -o qemu/security_features_flash_image.bin -f raw merge-args ../qemu/qemu_flash_args"...
usage: esptool merge_bin [-h] --output OUTPUT [--format {raw,uf2,hex}]
                         [--chunk-size CHUNK_SIZE] [--md5-disable]
                         [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                         [--flash_mode {keep,qio,qout,dio,dout}]
                         [--flash_size {keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                         [--spi-connection SPI_CONNECTION]
                         [--target-offset TARGET_OFFSET]
                         [--fill-flash-size {256KB,512KB,1MB,2MB,4MB,8MB,16MB,32MB,64MB,128MB}]
                         <address> <filename> [<address> <filename> ...]
esptool merge_bin: error: argument <address> <filename>: Address "merge-args" must be a number
merge_bin failed with exit code 2, output of the command is in the /home/<user>/SW/esp/test_projects/security_features_app/build/log/idf_py_stderr_output_39201 and /home/<user>/SW/esp/test_projects/security_features_app/build/log/idf_py_stdout_output_39201
```

The content of the `../qemu/qemu_flash_args` file is not being expanded to the command line, instead the string `merge-args ../qemu/qemu_flash_args` is just passed as such to the esptool, who doesn't know what to do with it.

If leaving out the `merge-args ../qemu/qemu_flash_args` part, the default `build/flash_args` file will be used [instead](https://github.com/espressif/esp-idf/blob/master/tools/idf_py_actions/serial_ext.py#L314). This produces  a binary flash image.

[... 28 lines cut ...]
[1/4] cd /home/<user>/SW/esp/test_projects/security_features_app/build/esp-idf/esptool_py && /home/<user>/.espressif/pytho...artition_table/partition-table.bin /home/<user>/SW/esp/test_projects/security_features_app/build/security_features_app.bin
security_features_app.bin binary size 0x41000 bytes. Smallest app partition is 0x100000 bytes. 0xbf000 bytes (75%) free.
[1/2] cd /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/esp-idf/esptool_py && /home/<user>/.espr....py --offset 0xd000 bootloader 0x0 /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/bootloader.bin
Bootloader binary size 0xb000 bytes. 0x2000 bytes (15%) free.
[2/2] cd /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/esp-idf/esptool_py && /home/<user>/.espr....py --offset 0xd000 bootloader 0x0 /home/<user>/SW/esp/test_projects/security_features_app/build/bootloader/bootloader.bin
Bootloader binary size 0xb000 bytes. 0x2000 bytes (15%) free.
[4/4] Completed 'bootloader'Executing action: qemu
Using provided flash image: build/qemu/security_features_flash_image.bin
Using existing efuse image: qemu_efuse.bin
Running qemu (fg): qemu-system-xtensa -M esp32s3 -drive file=build/qemu/security_features_flash_image.bin,if=mtd,format=raw -drive file=qemu_efuse.bin,if=none,format=raw,id=efuse -global driver=nvram.esp32c3.efuse,property=drive,value=efuse -global driver=timer.esp32s3.timg,property=wdt_disable,value=true -nic user,model=open_eth -nographic -serial mon:stdio
Adding SPI flash device
qemu-system-xtensa: Drive size error: only 2, 4, 8, and 16MB images are supported
qemu-system-xtensa: -drive file=build/qemu/security_features_flash_image.bin,if=mtd,format=raw: machine type does not support if=mtd,bus=0,unit=0
Done
```

Either the instructions are wrong or I have something setup incorrectly in my environment. If I just run `idf.py qemu` or `idf.py qemu monitor`, the generated image is valid and it runs just fine in Qemu (at least the first boot where it goes through the init stuff and then resets). I.e. my Qemu installation etc. are okay. The problem is somehow with the final size of the generated image that contains the encrypted partitions etc.

Side-note: I used the `--efuse-file qemu_efuse.bin ` option for the Qemu run commands, since I noticed that the `build/qemu_efuse.bin` file gets rebuilt every-time you have "burned" a fuse in it. I.e. the content would keep being reset and it was thus not possible to follow all the instructions in the example. By placing it outside the build directory, it would remember the burnt eFuses.

### Build or installation Logs.

```plain

```

o-marshmallow, 2025-08-20:
Hello @friissoren ,

Can you try this commnad instead:
```
cd build
esptool.py --chip esp32s3 merge_bin --fill-flash-size 4MB -o flash_image.bin @flash_args
```
Change the `chip` argument arccordingly, the QEMU-related documentation can be found here: https://github.com/espressif/esp-toolchain-docs/blob/main/qemu/esp32s3/README.md

Ashish285, 2025-08-20:
Sure, thanks for confirming this. I will update the documentation and include in the support to keep the QEMU eFuses out of the build directory. On a rebuild, it may get changed and require the eFuses to be flashed again.

closed 2025-08-26 by espressif-bot