espressif/esp-idf #17460: override_freq_hz is used unititialized- causes SPI frequency to peg at APB clock (80MHz)

FurTrader, 2025-08-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50500.0 (5.5.0)

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

I have an OLED display on SPI3, using u8g2. The SPI frequency should be 10MHz, as set by `spi_device_interface_config_t.clock_speed_hz   = 10000000`

### What is the actual behavior?

After updating to ESP-IDF 5.5.0, the SPI frequency is 80MHz, and the OLED does not work (the waveform is distorted- too high for this board)

### Steps to reproduce.

call `spi_device_transmit()` WITHOUT initializing `spi_transaction_t.override_freq_hz` (the param did not exist prior to 5.5.0)

If the uninitialized param contains garbage greater than APB frequency, the frequency that was already set by `spi_device_interface_config_t.clock_speed_hz` will be overridden by `spi_ll_master_cal_clock` and pegged at the APB clock frequency.

### Debug Logs.

```plain
E (3918) override_freq_hz: 1070267208

This was logged in my setup code, without initializing trans_desc.override_freq_hz.
It shows that this param is passing stack garbage to spi_device_transmit().

ESP_LOGE("override_freq_hz", "%lu", trans_desc.override_freq_hz);
```

### Diagnostic report archive.

[diag-4d5bf598-9a7f-4b0a-9170-80c1d66a08fb.zip](https://github.com/user-attachments/files/21863284/diag-4d5bf598-9a7f-4b0a-9170-80c1d66a08fb.zip)

### More Information.

The compiler did not give a warning that the param is used uninitialized.

Without initializing `override_freq_hz`, any call to `spi_device_transmit()` will peg the SPI clock to the APB clock, breaking existing code if the hardware can't support that frequency.

I use Vscode and platformio on windows 11.

FurTrader, 2025-08-19:
this change was introduced in Commit 39c8526

FurTrader, 2025-08-20:
AH! So simple! This began as library code and I didnt notice. (or understand, really- I'm a novice at best)

I will update my code to this style.

I still think its a bug that the driver defaults to the maximum clock speed, instead of `spi_device_interface_config_t.clock_speed_hz`

wanckl, 2025-08-20:
@FurTrader `spi_transaction_t.override_freq_hz` is user config param, you should response for init it, driver can only add some validation check

FurTrader, 2025-08-20:
Sure, but why default to maximum speed if the input is invalid? This is likely to break hardware.