espressif/esp-idf #17466: esp_ringbuf: Buffer consistency damaged when returning single item in buffer full condition

KlausPopp, 2025-08-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5 8c750b088c7cd857d079c0eeb495da199b359461

### Espressif SoC revision.

ESP32 (Rev 3)

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Any ESP32, but also Linux host

### Power Supply used.

USB

### What is the expected behavior?

In a no-split ringbuffer: When ringbuffer is full, and the first item is returned, only the first item should be free'd.

### What is the actual behavior?

In a no-split ringbuffer: When ringbuffer is full, and the first item is returnedthe whole buffer i.e. all items are free'd.

### Steps to reproduce.

* Create a no-split ring buffer with 4 items.
* Fill all items with Acquire and Send
* Call Receive for all items
* Return the first item
* Verify that the first item is returned correctly and only the first item
--> After that, all items are freee

See this test case: https://github.com/ci4rail/esp-idf/blob/ca663e6caad5ab63f05ed44b8036ed7f6480146e/components/esp_ringbuf/test_apps/main/test_ringbuf_common.c#L1243

### Debug Logs.

```plain

```

### More Information.

Fix is available here: https://github.com/ci4rail/esp-idf/commit/ca663e6caad5ab63f05ed44b8036ed7f6480146e

KonstantinKondrashov, 2025-08-20:
Regarding the assertion you posted: it indicates the item you pass to vRingbufferReturnItem() is already marked as returned. The most common cause is calling vRingbufferReturnItem() twice for the same pointer (or passing a pointer that doesn’t exactly match the one returned by xRingbufferReceive()). Please double-check your code: call vRingbufferReturnItem() exactly once per successful xRingbufferReceive().

Regarding [the test](https://github.com/ci4rail/esp-idf/blob/ca663e6caad5ab63f05ed44b8036ed7f6480146e/components/esp_ringbuf/test_apps/main/test_ringbuf_common.c#L1281-L1284). `uxFree` is not “free bytes”. Per the docs, it’s the free pointer position (an offset inside the buffer), just like uxRead/uxWrite/uxAcquire are pointer positions—not sizes.
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/freertos_additions.html
So this check is wrong: `TEST_ASSERT_EQUAL(ITEM_HDR_SIZE + MEDIUM_ITEM_SIZE, uxFree);`
If your buffer storage starts at offset 0 and you return the first item, the start of the free region (the free pointer) will typically be at offset 0, not at “size of the returned item”. You’re comparing a pointer (offset) with a size.

`xRingbufferGetCurFreeSize` returns a free size.

KlausPopp, 2025-08-21:
Hello Konstantin, 

thanks for your fast reply. Before discussing about my testcase, let's focus on the functional behavior, which I think is wrong:

I repeat my scenario:

* Create a no-split ring buffer with 4 items (items 0..3).
* Producer: Fill all items with Acquire and Send
* Consumer: Call Receive for all items
* Consumer: Return item 0

    --> After that, all items are free. 

This is not the expected behaviour. 

The consumer only returned item 0, so he thinks he still owns items 1..3.
But because returning item 0 causes all items to be freed, they are now again available to the producer as well. If the producer makes further acquire calls, the items are then owned by the producer and the consumer at the same time, so the buffer management is corrupted. After such a situation, I have seen the crash mentioned above.

The root cause is here: https://github.com/espressif/esp-idf/blob/856bba66bc751b4cfe53f1f4bba53fa291bae453/components/esp_ringbuf/ringbuf.c#L656 

```
static void prvReturnItemDefault(Ringbuffer_t *pxRingbuffer, uint8_t *pucItem)
{
        ....
        } else if (pxRingbuffer->pucFree == pxRingbuffer->pucAcquire && pxRingbuffer->pucFree == pxRingbuffer->pucRead) {
            //Special case where a full buffer is completely freed in one go
            pxRingbuffer->uxRingbufferFlags &= ~rbBUFFER_FULL_FLAG;
        }
```

This causes all items to be freed when in the described scenario. 

Do you agree that the current behavior is wrong in the describes scenario?

KonstantinKondrashov, 2025-08-21:
Thanks for the explanation. Yes, I agree. I observe the same behavior on my side, too. After we return item 0, the full buffer is marked as free and available for xRingbufferSendAcquire again, even though not all items have been returned/read.

It is bug the doc says - [.... . However, the freeing of space must occur in FIFO order, therefore not returning the earliest retrieved item prevents the space of subsequent items from being freed.](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/freertos_additions.html#retrieving-returning).

AxelLin, 2025-09-15:
@Alvin1Zhang 
All branches do not update recently, this seems unusual. especially, v5.2 branch does not update for more than 1 Month.

closed 2025-09-17 by espressif-bot