espressif/esp-idf #17482: secure_boot_v2 is using MMU to map SBV2 footer from flash, bad content read (SOLVED)

g-380, 2025-08-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.5

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

DevKitC-1 : ESPRESSIF ESP32 S3-WROOM-1_N8R8 

### Power Supply used.

USB

### What is the expected behavior?

The application has downloaded a 'raw'-[especure-signed] binary (not an ESP image) writtten in custom DATA partition using https_ota. 
Public key hashed is configured.
We expect 'bootloader_support' compinent with esp_secure_boot_verify_signature function to validate the signature

### What is the actual behavior?

verification stopped when check magc number

### Steps to reproduce.

1. Sign the 'raw' binary with default sign key secure_boot_signing_key.pem using  espsecure.py sign-data --version 2 \
  --keyfilesecure_boot_signing_key.pem \
  --output signed-app.bin \
  build/pre_encrypted_ota.bin
and put it on https (local) download server
  
2 : patch static esp_err_t ota_verify_partition(ota_ops_entry_t *ota_ops) from esp_ota_ops.c

@ -470,9 +488,23 @@ static esp_err_t ota_verify_partition(ota_ops_entry_t *ota_ops)
                 return ESP_ERR_OTA_VALIDATE_FAILED;
             }
             esp_partition_munmap(partition_table_map);
-        }
-    }
-    return ret;
+        }        
+       }
+       //treate data partition as raw signed binary, so map the Fotter Sig SBV-2 to verify sig
+      else if (ota_ops->partition.final->type == ESP_PARTITION_TYPE_DATA) {
+                  esp_err_t e = esp_secure_boot_verify_signature((uint32_t)ota_ops->partition.staging->address,  ALIGN_DOWN(ota_ops->wrote_size, SPI_FLASH_SEC_SIZE) - SPI_FLASH_SEC_SIZE);
+                       if (e != ESP_OK) {
+                               ESP_LOGE(TAG,"esp_secure_boot_verify_signature KO");
+                               return ESP_ERR_OTA_VALIDATE_FAILED;
+                       }
+                       return ESP_OK;
+               }

3. idf.py build then flash on esp32s3
=> pre_encrypted_ota_example: esp_secure_boot_verify_signature KO: ESP_ERR_INVALID_RESPONSE (0x108)

### Debug Logs.

```plain
With workaround 1 (see above), memory is mapped from Flash accordingly, SIV SBV2 footer can be used
=> then, digest is found for public key (OK) and crypto verification fail on DIGEST (see workaound 2 above)

OK CASE : 
I (8932) secure_boot_v2: Take trusted digest key(s) from eFuse block(s)
I (8942) secure_boot_v2: validate_signature_block
I (8942) secure_boot_v2: Generate the SHA of the public key components in the signature block
I (8952) secure_boot_v2: bootloader_sha256_finish
I (8952) secure_boot_v2: #0 app key digest == #0 trusted key digest
I (8962) secure_boot_v2: Verifying with RSA-PSS...
I (9022) secure_boot_v2_rsa: Signature verified successfully!
I (9022) pre_encrypted_ota_example: ESP_HTTPS_OTA upgrade successful. Rebooting ...
```

### More Information.

Workaround 1 
-    const ets_secure_boot_signature_t *sig_block = bootloader_mmap(src_addr + padded_length, sizeof(ets_secure_boot_signature_t));
-    if (sig_block == NULL) {
-        ESP_LOGE(TAG, "Failed to mmap data at offset 0x%"PRIx32, src_addr + padded_length);
-        return ESP_FAIL;
-    }

map the flash with esp_flash_read to get the block 

static uint8_t sig_sector[SPI_FLASH_SEC_SIZE];  
+                                err = esp_flash_read(esp_flash_default_chip,
+                                                                                          sig_sector,
+                                                                                          src_addr + padded_length,
+                                                                                          sizeof(sig_sector));

+ const ets_secure_boot_signature_t *block =
+                                       (const ets_secure_boot_signature_t *)sig_sector;
+                
+                               err = esp_secure_boot_verify_sbv2_signature_block(sig_block, digest, NULL);

Workaround 2
Same job for SHA256 : replace bootloader_sha256_flash_contents with SPI flash map access

mahavirj, 2025-08-28:
@g-380 

Is it possible for you to supply a small application project to recreate the issue?

g-380, 2025-08-29:
hi @mahavirj  , OK I will do that

Ashish285, 2025-09-03:
Hi @g-380 ,

I have tried to replicate this issue on my end but I am unable to do so. Can you please help to verify if I missed anything here:
1. I used pre_encrypted_ota example as base and enabled Secure Boot V2 + Flash encryption (in development mode)
2. I added in secure boot verification for `ESP_PARTITION_TYPE_DATA` as you suggested in the issue
3. I used a storage binary (which is not ESP generated), signed it using `espsecure.py` and then encrypted using `esp_enc_img_gen.py`
<details><summary>espsecure.py</summary>
<p>

espsecure.py sign_data --version 2 --keyfile secure_boot_signing_key.pem --output signed_storage.bin ./storage.bin

</p>
</details> 

<details><summary>esp_enc_img_gen.py</summary>
<p>

python ../../tools/esp_enc_img_gen.py encrypt build/signed_storage.bin rsa_key/private.pem build/signed_storage_encrypted.bin

</p>
</details> 

4. I then copied this binary over to the server.
6. On the pre_encrypted_ota example, I made the following change to allow writing the storage binary to DATA partition.
```
diff --git a/esp_encrypted_img/examples/pre_encrypted_ota/main/pre_encrypted_ota.c b/esp_encrypted_img/examples/pre_encrypted_ota/main/pre_encrypted_ota.c
index 1dba7ab..5aa3454 100644
--- a/esp_encrypted_img/examples/pre_encrypted_ota/main/pre_encrypted_ota.c
+++ b/esp_encrypted_img/examples/pre_encrypted_ota/main/pre_encrypted_ota.c
@@ -174,6 +174,11 @@ void pre_encrypted_ota_task(void *pvParameter)
         .enc_img_header_size = esp_encrypted_img_get_header_size(),
     };

+    ota_config.partition.staging = NULL; // free app ota partition will be selected and used for downloading a new image
+    ota_config.partition.final = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "storage");
+    assert(ota_config.partition.final != NULL);
+    ota_config.partition.finalize_with_copy = true;

```

7. I can see that Secure Boot V2 is enabled, Flash encryption is enabled and the Secure Boot verification also succeeds for the image.

```
I (379) esp_image: Verifying image signature...
I (380) secure_boot_v2: Verifying with RSA-PSS...
I (383) secure_boot_v2: Signature verified successfully!
I (392) boot: Loaded app from partition at offset 0x20000
I (429) boot: Set actual ota_seq=1 in otadata[0]
I (429) secure_boot_v2: enabling secure boot v2...
I (429) secure_boot_v2: secure boot v2 is already enabled, continuing..
I (433) boot: Checking flash encryption...
I (437) flash_encrypt: flash encryption is enabled (0 plaintext flashes left)
...
I (5055) esp_https_ota: Starting OTA...
I (5055) esp_https_ota: Writing to <ota_1> partition at offset 0x1d0000
I (5055) esp_ota_ops: Staging partition - <ota_1>. Final partition - <storage>.
I (5075) esp_encrypted_img: Magic Verified
I (5075) esp_encrypted_img: Reading RSA private key
I (6495) pre_encrypted_ota_example: Running firmware version: 904b50a-dirty
I (6585) secure_boot_v2: Take trusted digest key(s) from eFuse block(s)
I (6585) secure_boot_v2: #0 app key digest == #0 trusted key digest
I (6585) secure_boot_v2: Verifying with RSA-PSS...
I (6645) secure_boot_v2_rsa: Signature verified successfully!
I (6645) esp_ota_ops: esp_secure_boot_verify_signature OK
I (6645) esp_ota_ops: Copy from <ota_1> staging partition to <storage>...
I (6715) pre_encrypted_ota_example: ESP_HTTPS_OTA upgrade successful. Rebooting ...
```

Edit:

It seems like in your `partitions.csv` you have not marked the `ota_2, data` partition to be encrypted. According to the [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#encrypted-partitions) not all the partitions are encrypted by default and `data` partition needs the `encrypted` flag to be set.

`ota_2,     data, 0x40,             ,2M, encrypted`

Can you please try with this if it solves the issue? We will be releasing a fix so that unencrypted partitions can also be verified if secure boot and flash encryption is enabled.

g-380, 2025-09-04:
Hi @Ashish285 

You're dam right, what a mistake ! 
I will confirm but I have no doubt now...

could you explain ? : 

=> with SPI flash access, I read OTA 2 plaintext so it works 

=> what about MMU access and cache content ?

Ashish285, 2025-09-04:
Hi @g-380 ,

Please have a look at the [docs for Memory Synchronization](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/mm.html#memory-synchronization). If `esp_flash` driver APIs are used, it handles the memory synchronisation internally.