espressif/esp-idf #17483: WiFi registers the STA/AP interface twice occasionally

atanisoft, 2025-08-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.2 release tag

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I'd expect to be able to connect WiFi to a random station, disconnect using `esp_wifi_stop()`, call `esp_wifi_set_mode(WIFI_MODE_APSTA)` and trigger a non-blocking scan via `esp_wifi_scan_start` using the default config. 

### What is the actual behavior?

Actual behavior is the first time it works but running the same sequence of steps a second time reports a failure:
```
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 2
WiFi station started, initiating connection.
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 4
Reloading mDNS entries for station interface
Event received: IP_EVENT / 0
Exiting as complete
heap:72.65kB (max block:30.99kB), PSRAM:1887.77kB (max block:1855.99kB), mainBufferPool:0.16kB, tasks:12
Allocating new alias D4F for node 0202050311e1
[mDNS] Found nexxtcs providing service:_openlcb-can._tcp on port 12021
Allocating new alias D4F for node 0202050311e1
Invoking train search request 1 for address 3
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 5
Skipping reconnection attempt due to configuration.
Event received: WIFI_EVENT / 3
GCHubPort: Shutting down gridconnect port -1. (0x3c218d68)
Disconnecting WiFi
calling init_common
setting WIFI_MODE_APSTA
setting WIFI_STORAGE_RAM
calling esp_wifi_start
calling esp_wifi_set_max_tx_power
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 2
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 12
Event received: WIFI_EVENT / 1
Scan status: 0, 19 SSIDs
... SSIDs ...
heap:66.78kB (max block:28.99kB), PSRAM:1877.65kB (max block:1855.99kB), mainBufferPool:4.40kB, tasks:12
Event received: WIFI_EVENT / 3
Event received: WIFI_EVENT / 13
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 2
WiFi station started, initiating connection.
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 4
Reloading mDNS entries for station interface
Event received: IP_EVENT / 0
Exiting as complete
Searching for command station.
[mDNS] Found nexxtcs providing service:_openlcb-can._tcp on port 12021
Command station found, connecting.
Connecting to 192.168.4.1:12021
Allocating new alias D4F for node 0202050311e1
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 5
Skipping reconnection attempt due to configuration.
Event received: WIFI_EVENT / 3
GCHubPort: Shutting down gridconnect port -1. (0x3c21718c)
heap:68.67kB (max block:30.99kB), PSRAM:1881.05kB (max block:1855.99kB), mainBufferPool:4.40kB, tasks:12
Disconnecting WiFi
calling init_common
setting WIFI_MODE_APSTA
setting WIFI_STORAGE_RAM
calling esp_wifi_start
calling esp_wifi_set_max_tx_power
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 2
Event received: WIFI_EVENT / 43
Event received: WIFI_EVENT / 12

assert failed: netif_add /IDF/components/lwip/lwip/src/core/netif.c:420 (netif already added)

Backtrace: 0x4037a51d:0x3fcdb310 0x40383039:0x3fcdb330 0x4039124a:0x3fcdb350 0x420b67fb:0x3fcdb490 0x4212b5fa:0x3fcdb4c0 0x4212b1d2:0x3fcdb520 0x4038f849:0x3fcdb540
--- 0x4037a51d: panic_abort at /trains/esp-idf-v5.4.2/components/esp_system/panic.c:469
--- 0x40383039: esp_system_abort at /trains/esp-idf-v5.4.2/components/esp_system/port/esp_system_chip.c:87
--- 0x4039124a: __assert_func at /esp-idf-v5.4.2/components/newlib/assert.c:80
--- 0x420b67fb: netif_add at /esp-idf-v5.4.2/components/lwip/lwip/src/core/netif.c:420
--- 0x4212b5fa: esp_netif_lwip_add at /esp-idf-v5.4.2/components/esp_netif/lwip/esp_netif_lwip.c:921
---  (inlined by) esp_netif_start_api at /esp-idf-v5.4.2/components/esp_netif/lwip/esp_netif_lwip.c:1105
--- 0x4212b1d2: esp_netif_api_cb at /esp-idf-v5.4.2/components/esp_netif/lwip/esp_netif_lwip.c:200
--- 0x4038f849: tcpip_thread_handle_msg at /esp-idf-v5.4.2/components/lwip/lwip/src/api/tcpip.c:207
---  (inlined by) tcpip_thread at /esp-idf-v5.4.2/components/lwip/lwip/src/api/tcpip.c:148
```

### Steps to reproduce.

Connect to a station, disconnect, scan and configure for a station (can be same one), disconnect, trigger scan and there will be a failure.

Note it is a private set of projects but I can share the code which is being used and reproduces this.

### Debug Logs.

```plain

```

### Diagnostic report archive.

[diag.zip](https://github.com/user-attachments/files/21944395/diag.zip)

### More Information.

Very similar behavior to https://github.com/espressif/esp-idf/issues/8702 but seen on v5.4.2 under very certain circumstances.

hansw123, 2025-09-02:
@atanisoft  sorry for late reply
could you provide the mini action that can reproduce this issue? thanks

hansw123, 2025-09-08:
@atanisoft 
maybe you can post your code and config in here

atanisoft, 2026-05-15:
@hansw123 I am able to reproduce this crash at-will every time with ESP-IDF 5.4.4 now, I've also look at 6.0 but it hasn't changed there either. I need to know how to work around this issue since there doesn't seem to be activity from Espressif on fixing it. It is quite odd that it wants to register an interface twice when it should only be done once. I can share privately an ESP32-S3 binary of the app that reproduces this easily.

kyostio, 2026-07-17:
For me, the host's (=esp32-p4) `esp_hosted` and the corresponding firmware on the c6 chip were not in sync. Once both were using the same version, the wifi started to work. In my case, the wifi didn't work at all, it crashed right out of the box every time.

I used this project to flash the c6 firmware: https://github.com/lboshuizen/crowpanel-p4-c6-sdio-ota as I'm not sure if I have a cable to flash the c6 firmware. afaik, it's not accessible via the USB connected to the P4 chip.