espressif/esp-idf #17484: Possible breaking change in 5.5 esp_wifi_set_config

hachi, 2025-08-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This change ce3611db265d3fa96e21834bf9e248d5e0fde8de which I believe was applied to 5.5 as 5558a08c07 has changed the semantics of esp_wifi_set_config() to not work while connecting.

Note: I have no issue with this change, I am only raising this issue because it was not noted in the 5.5 release notes as a breaking change.

Prior to 5.5 I believe this interface was unreliable when called if wifi was connecting, possibly not taking effect till cycled, but there are example and existing code that can end up stuck if a wifi SSID or password do not connect properly.

It seems the intended behavior is to call esp_wifi_disconnect() before calling this function, or wait till WIFI_EVENT_STA_BEACON_TIMEOUT is fired? I'm unsure.

I have specifically noticed this because of a ble provisioning application test where I end up receiving a log message saying I cannot set_config while connecting, and if the values are being read from NVS during startup I can't reboot to clear the situation.

hachi, 2025-08-27:
I have tested 5.4, 5.4.1 and 5.4.2 and this behavior is new in 5.5.

This is probably important to look at, is anyone reading this?

hachi, 2025-08-27:
As I said, the change is not the problem... the problem is that this change was not noted in the release notes, and it is a breaking change.

hansw123, 2025-08-28:
@hachi 
already add breaking change notes in **https://github.com/espressif/esp-idf/releases/tag/v5.5**

hachi, 2025-08-28:
Thank you for adding it there, but it is also missing here: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.5/wifi.html