espressif/esp-idf #17492: Missing warnings about functions with missing return directives (C++)

Smity, 2025-08-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

If a function has a return type but not all code paths lead to a return statement, the compilation should fail with an error such as `error: control reaches end of non-void function [-Werror=return-type]`. This is the observed behaviour in C code, but not all C++ code.

### What is the actual behavior?

IDF builds the function just fine and mayhem ensues at runtime. Usually, I find the bug in the form of a stack corruption; though in the following case, omitting the return statement causes `i` to continue incrementing while the loop runs infinitely:

```
int example()
{
    for (int i = 0; i < 32; i++)
    {
        printf("i = %i\n", i);
    }
}
```

### Steps to reproduce.

1. Make a function
2. Omit the `return` statement
3. Enjoy the show

### Build or installation Logs.

```plain

```

igrr, 2025-08-25:
Could you please attach a source file which demonstrates this issue, as well as the compile_commands.json file from your build directory?

(Other similar issues on the same topic ended up being related to one of the components in the project disabling this warning, e.g. https://github.com/espressif/esp-idf/issues/17006)

Smity, 2025-08-28:
I don't have the original source file anymore, but I am able to replicate the issue in main.cpp (or anywhere else) using the code above. I did test that code before submitting the bug report to make sure it wasn't actually something I'd done elsewhere. 

We've also discovered the issue can also be replicated by calling something like `noreturnB` (I put it in a loop here, but it doesn't have to be). Calling `noreturnB` in a loop prints `i` and some other messages from earlier in app_main, but `i` does _not_ increment. If I remove the code in app_main before the loop the output is diffrent (see below), but I expect the cause is the same in each case—jumping (or not jumping) to the wrong location.

I also tried these examples without looping and got pretty similar results.

```
int noreturnB()
{
    static int i = 0;
    printf("i = %i; ", i);
    i++;
}

extern "C" void app_main(void)
{
    if (esp_flash_encryption_enabled())
        ESP_LOGI(__func__, "Flash Encryption: ENABLED");
    else
        ESP_LOGI(__func__, "Flash Encryption: DISABLED");

    if (esp_secure_boot_enabled())
        ESP_LOGI(__func__, "Secure Boot: ENABLED");
    else
        ESP_LOGI(__func__, "Secure Boot: DISABLED");

    auto rst_reason = esp_reset_reason();
    ESP_LOGE("Reset Reason", "%s", RESET_REASON_STR[rst_reason]);

    for (int i = 0; i < 32; i++)
    {
        int j = noreturnB();
        printf("j = %i\n", j);
    }

    vTaskDelete(nullptr);
}
```

Output from the code above (boot logs removed):
```
...
I (895) app_main: Flash Encryption: DISABLED
I (901) app_main: Secure Boot: DISABLED
E (905) Reset Reason: ESP_RST_POWERON
i = 0; I (909) app_main: Flash Encryption: DISABLED
I (915) app_main: Secure Boot: DISABLED
E (919) Reset Reason: ESP_RST_POWERON
i = 0; I (923) app_main: Flash Encryption: DISABLED
I (929) app_main: Secure Boot: DISABLED
E (933) Reset Reason: ESP_RST_POWERON
i = 0; I (937) app_main: Flash Encryption: DISABLED
I (943) app_main: Secure Boot: DISABLED
E (947) Reset Reason: ESP_RST_POWERON
... etc
```

The output from the calling `noreturnB` after removing the code preceding it (boot logs removed again):
```
...
I (876) main_task: Calling app_main()
start test
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4201dd0d  PS      : 0x00060230  A0      : 0x8201dd00  A1      : 0x3fcb58a0
--- 0x4201dd0d: ShiftRegister::~ShiftRegister() at C:/Repos/Program/components/ShiftRegister/ShiftRegister.cpp:26
A2      : 0x00000007  A3      : 0x00000000  A4      : 0xfffffff7  A5      : 0x3c103592
A6      : 0x3fcb58ac  A7      : 0x3fca6818  A8      : 0xcdd7a200  A9      : 0x3fcb5560
A10     : 0x00000007  A11     : 0x00000000  A12     : 0xfffffffe  A13     : 0x3c104b08
A14     : 0x00000002  A15     : 0x3fcb5770  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x0000000f  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffd
--- 0x400556d5: strlen in ROM
--- 0x400556e5: strlen in ROM

Backtrace: 0x4201dd0a:0x3fcb58a0 0x4201dcfd:0x3fcb58d0 0x42011f39:0x3fcb5900 0x420ee26f:0x3fcb5930 0x40389976:0x3fcb5970
--- 0x4201dd0a: ShiftRegister::~ShiftRegister() at C:/Repos/Program/components/ShiftRegister/ShiftRegister.cpp:24
--- 0x4201dcfd: noreturnB() at C:/Repos/Program/main/loops.cpp:14
--- 0x42011f39: app_main at C:/Repos/Program/main/main.cpp:184
--- 0x420ee26f: main_task at C:/Users/Smity/esp/v5.4.1/esp-idf/components/freertos/app_startup.c:208
--- 0x40389976: vPortTaskWrapper at C:/Users/Smity/esp/v5.4.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

Here are my compile commands. -Wall appears to be there.
[compile_commands.json](https://github.com/user-attachments/files/22016232/compile_commands.json)

closed 2025-11-27 by Alvin1Zhang