espressif/esp-idf #17493: Cannot deinitialize then reinitialize nimBLE HID without crashing

RoboticWorx, 2025-08-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello! I am developing a product with ESP32-C5 on ESP-IDF v5.5. I'm having an issue where I am unable to deinitialize and then reinitialize Bluetooth nimBLE as a HID device without crashing with what is usually a Guru Meditation Error.

On first initialization everything works fine, then I can deinitialize fine to preserve power until the user wants to use Bluetooth again, but if I initialize again after that I get the crash so the system has to reboot before I can successfully initialize Bluetooth again. I got most my info from the esp_hid_device example, but there is nothing in there about deinitializing Bluetooth to use again later. Being able to deinitialize it completely is important to my application to save power since it could be a long time until the user wants to use Bluetooth again. Any help would be greatly appreciated!

Below is my full log. I am aware it says it is already initialized, but I have been trying many different things for days and even if I implement my own hid_gap_deinit function I will always get some sort of crash or problem when reinitializing. Most of the time this is a Guru Meditation Error if I remove ESP_ERROR checks.

`I (11412) BLE_INIT: Using main XTAL as clock source
I (11412) BLE_INIT: ble controller commit:[35fe65f]
I (11412) BLE_INIT: Bluetooth MAC: d0:cf:13:e0:a7:2e
I (11412) phy_init: phy_version 102,171bf417,Jun 12 2025,15:57:12
I (12132) phy: libbtbb version: 09fb4d6, Jun 12 2025, 15:57:24
I (12142) NimBLE: GAP procedure initiated: stop advertising.

I (12142) NimBLE: GAP procedure initiated: stop advertising.

I (12142) NimBLE: GAP procedure initiated: advertise; 
I (12152) NimBLE: disc_mode=2
I (12152) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=48 adv_itvl_max=80
I (12162) NimBLE: 

I (12422) ESP_HID_GAP: connection established; status=0
I (12482) NimBLE: encryption change event; status=0 
I (12482) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=11 reason=3 prevn=0 curn=0 previ=0 curi=1

I (12482) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=21 reason=3 prevn=0 curn=1 previ=0 curi=0

I (12492) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=40 reason=3 prevn=0 curn=1 previ=0 curi=0

I (12502) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=47 reason=3 prevn=0 curn=1 previ=0 curi=0

I (12512) ESP_HID_GAP: mtu update event; conn_handle=0 cid=4 mtu=256
I (14932) NimBLE: GAP procedure initiated: stop advertising.

I (14932) NimBLE: GAP procedure initiated: stop advertising.

I (14932) NimBLE: GAP procedure initiated: terminate connection; conn_handle=0 hci_reason=19

I (14952) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=11 reason=2 prevn=0 curn=0 previ=1 curi=0

I (14952) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=21 reason=2 prevn=1 curn=0 previ=0 curi=0

I (14962) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=40 reason=2 prevn=1 curn=0 previ=0 curi=0

I (14972) ESP_HID_GAP: subscribe event; conn_handle=0 attr_handle=47 reason=2 prevn=1 curn=0 previ=0 curi=0

I (14982) ESP_HID_GAP: disconnect; reason=534
I (14992) BLUETOOTH_FUNCS: Bluetooth fully disabled
E (18012) ESP_HID_GAP: Already initialised
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4206cc4c
--- 0x4206cc4c: bluetooth_init at C:/Projects/ESP/PolyCast5/components/bluetooth/src/bluetooth_funcs.c:903
file: "./components/bluetooth/src/bluetooth_funcs.c" line 903
func: bluetooth_init
expression: ret

abort() was called at PC 0x408152ad on core 0
--- 0x408152ad: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_system/esp_err.c:49
Core  0 register dump:
MEPC    : 0x408008b0  RA      : 0x408152b8  SP      : 0x40856cf0  GP      : 0x40823304
--- 0x408008b0: panic_abort at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_system/panic.c:483
--- 0x408152b8: __ubsan_include at C:/Espressif/frameworks/esp-idf-v5.5/components/esp_system/ubsan.c:311
TP      : 0x40856ff0  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000004  S1      : 0x40856d54  A0      : 0x40856d1c  A1      : 0x40856d52
[... 104 lines cut ...]
void bluetooth_deinit(void)
{
	ble_gap_adv_stop(); // Stop advertising

	if (nimble_port_stop()) {
		ESP_LOGE(TAG, "nimble_port_stop failed");
		return;
	}

	if (nimble_port_deinit() != ESP_OK) {
		ESP_LOGE(TAG, "nimble_port_deinit failed");
		return;
	}

	if (ble_hid_param.hid_dev) {
		esp_hidd_dev_deinit(ble_hid_param.hid_dev);
		ble_hid_param.hid_dev = NULL;
	}

	#ifdef POLYCAST5_DEBUG
		ESP_LOGI(TAG, "Bluetooth fully disabled");
	#endif
}`

Thank you for your time and assistance. This has been driving me crazy for awhile. If there is a easy fix for this is may be helpful for others to see a deinit function in the example: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_hid_device

linkua, 2025-08-26:
"The esp_hid_device_demo using a Nimble keyboard to connect to iOS does not work！"

I (947936) HID_DEV_BLE: CONNECT
I (947936) ESP_HID_GAP: connection established; status=0
I (948266) ESP_HID_GAP: mtu update event; conn_handle=1 cid=4 mtu=256
I (948506) ESP_HID_GAP: subscribe event; conn_handle=1 attr_handle=8 reason=1 prevn=0 curn=0 previ=0 curi=1

I (949526) ESP_HID_GAP: subscribe event; conn_handle=1 attr_handle=37 reason=1 prevn=0 curn=1 previ=0 curi=0

rahult-github, 2025-08-26:
Hi @RoboticWorx  ,

For your issue , can you please do below changes and check if it helps. 

Step1 . Apply attached patch in $IDF_PATH

[0001-fix-nimble-Add-check-for-hid-gap-deinit.txt](https://github.com/user-attachments/files/21986131/0001-fix-nimble-Add-check-for-hid-gap-deinit.txt)
 

Step2, In your application, change your bluetooth_deinit function to include calls to esp_hid_gap_deinit and ble_gatts_reset as show below: 

`
void bluetooth_deinit(void)
{
        esp_hid_gap_deinit();

       ble_gap_adv_stop(); // Stop advertising

       ble_gatts_reset();

       if (nimble_port_stop()) {
               ESP_LOGE(TAG, "nimble_port_stop failed");
               return;
       }

       if (nimble_port_deinit() != ESP_OK) {
               ESP_LOGE(TAG, "nimble_port_deinit failed");
               return;
       }

       if (s_ble_hid_param.hid_dev) {
               esp_hidd_dev_deinit(s_ble_hid_param.hid_dev);
               s_ble_hid_param.hid_dev = NULL;
       }
}
`
And please check if this helps

rahult-github, 2025-08-26:
Hi @linkua  ,

For your issue, please use menuconfig to set NIMBLE_SM_LVL to 2. This is iphone side issue where it expects a particular security level to work.

RoboticWorx, 2025-08-28:
Hi @rahult-github 

Any thoughts on this?

Thank you,
Justin Atkins

strange969, 2025-08-29:
Hi @RoboticWorx ,

For the issues `Write on Protocol Mode Failed: 5` and `Unable to fetch protocol_mode`, could you please apply the below patch.

Apply attached patch in $IDF_PATH.
[fix_service_index_resetting.patch](https://github.com/user-attachments/files/22042241/fix_service_index_resetting.patch)

Kindly let us know if this resolves the problem.