espressif/esp-idf #17494: ULP_RISCV_CYCLES_PER_US links in floating point operations with significant code-size overhead

m-mcgowan, 2025-08-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The `ULP_RISCV_CYCLES_PER_US`define (and correspondingly `ULP_RISCV_CYCLES_PER_MS`, which is derived from it) declares the number of cycles using floating point constants:

```C

#define ULP_RISCV_CYCLES_PER_US 8.5
--
87 | #elif CONFIG_IDF_TARGET_ESP32S3
88 | #define ULP_RISCV_CYCLES_PER_US 17.5
89 | #endif
90 | #define ULP_RISCV_CYCLES_PER_MS ULP_RISCV_CYCLES_PER_US*1000
```
([source](https://github.com/espressif/esp-idf/blob/758939caecb16e5542b3adfba0bc85025517db45/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_utils.h#L86))

So, simple code like 

```C
ulp_riscv_delay_cycles(100 * ULP_RISCV_CYCLES_PER_US);
```

Results in *double-precision* floating point operations which are implemented in `__muldf3`, `__fixunsdfsi`, `__floatsidf`.

An `nm` dump of the ULP elf file shows these add considerable overhead:

```
00000086 00000440 T __muldf3
00000528 0000006e T __floatsidf
000004c6 00000062 T __fixunsdfsi
```
Size is given in the second column - summing these gives a a total of *1296 bytes added*.

## Suggested Fixes

The crux of the fix is of course to avoid floating point altogether, especially for something as trivial as computing a cycle count or implementing a delay. 

1.  Provide numerator and denominator values so fixed point arithmetic can be used.
2. Provide delay macros/functions implementing delays in microseconds and milliseconds. Having a delay function for milliseconds would help avoid any potential overflow relating to the scaling factor.

### Numerator and Denominator Values

Constants define the ratio of microseconds to cycle count using a numerator and denominator:
```C

// for the ESP32S3 (17.5 cycles per us)
#define ULP_RISCV_CYCLES_PER_US_NUM (35)
#define ULP_RISCV_CYCLES_PER_US_DENOM (2)
```

These compute the number of cycles using integer multiplication and division, avoiding floating point arithmetic.

For developer convenience, I suggest adding a macro, such as `ULP_RISCV_US_TO_CYCLES`:

```c
#define ULP_RISCV_US_TO_CYCLES(us) (us*ULP_RISCV_CYCLES_PER_US_NUM/ULP_RISCV_CYCLES_PER_US_DENOM)
```

Alternatively, the `ULP_RISCV_CYCLES_PER_US` could be changed to use the ratio:

```
#define ULP_RISCV_CYCLES_PER_US (ULP_RISCV_CYCLES_PER_US_NUM/ULP_RISCV_CYCLES_PER_US_DENOM)
```

Then existing API could be used without floating-point overhead:

```C
ulp_riscv_delay_cycles(100 * ULP_RISCV_CYCLES_PER_US);
```

New functions or macros that express delays in micro or milliseconds would provide a higher-level API:

```C
#define ulp_riscv_delay_us(us) ulp_riscv_delay_cycles((us)*ULP_RISCV_CYCLES_PER_US)
```

The delay functions are nuanced in their implementation, having to take into account the function call overhead (see  https://github.com/espressif/esp-idf/issues/16891). Having a higher-level API gives more freedom to the implementation to add improvements in accuracy over time.

m-mcgowan, 2025-08-26:
Applying these fixes locally, my ULP binary reduced from 3824 bytes to 2188 bytes, a difference of 1636 bytes - the floating point version is 74% larger than the fixed point version.  This represents a significant size saving for us, we need as much RTC SLOW RAM as we can get to buffer sensor data before waking up the MCU to process it.

KonstantinKondrashov, 2025-08-26:
Hi @m-mcgowan, thanks for raising this.
Originally, ulp_riscv_delay_cycles was implemented in a simple way, and I also had a fix prepared to remove the floating-point operation. That change wasn’t included in the original commit for [IDFGH-15960](https://github.com/espressif/esp-idf/issues/16891)
, but now that you’ve pointed it out, we can move forward with addressing it.

Your suggestion to switch from floating-point constants (ULP_RISCV_CYCLES_PER_US) to integer values makes sense and aligns with what we had in mind. Providing helper ulp_riscv_delay_us() would make the API more intuitive.

jrahlf, 2025-08-27:
how about using C23 constexpr?