espressif/esp-idf #17498: WiFi Mesh MESH_NODES swithcing to closer Parents if low RSSI

michaelsimp, 2025-08-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using ESP IDF ver 5.5.0 and WiFi Mesh network.
My network is self forming with some node having a weak RSSI to their parent, and I want to lower this.
IDF API provides 2 function for controlling thresholds but little by way of description on how to use them.
I used `esp_mesh_get_rssi_threshold(&threshold)` to read the default RSSI thresholds and received the following thresholds:

-             threshold.high = -78;
-             threshold.medium = -82;
-             threshold.low = -85;

So I lowered them using `esp_mesh_set_rssi_threshold(&threshold)` to:

-             threshold.high = -70;
-             threshold.medium = -75;
-             threshold.low = -80;

And similarly I used `esp_mesh_get_switch_parent_paras(&params)` to read the default params and got:

-             params.duration_ms = 60000; // Parent weak RSSI monitor duration, if the RSSI continues to be weak during this duration_ms, device will search for a new parent
-             params.cnx_rssi = -120;      // RSSI threshold for keeping a good connection with parent.
-             params.select_rssi = -78;   // RSSI threshold for parent selection. It should be a value greater than switch_rssi
-             params.switch_rssi = -78;   // Disassociate with current parent and switch to a new parent when the RSSI is greater than this set threshold
-             params.backoff_rssi = -78;  // RSSI threshold for connecting to the root

 So I lowered them using `esp_mesh_set_switch_parent_paras(&params)` to:

-             params.duration_ms = 60000; // Parent weak RSSI monitor duration, if the RSSI continues to be weak during this duration_ms, device will search for a new parent
-             params.cnx_rssi = -80;      // RSSI threshold for keeping a good connection with parent.
-             params.select_rssi = -70;   // RSSI threshold for parent selection. It should be a value greater than switch_rssi
-             params.switch_rssi = -70;   // Disassociate with current parent and switch to a new parent when the RSSI is greater than this set threshold
-             params.backoff_rssi = -68;  // RSSI threshold for connecting to the root

But some of my MESH_NODEs are still holding parent connections at -89dBm and are not switching to closer parent nodes.

The documents say thing like "select_rssi is the RSSI threshold for parent selection. It should be a value greater than switch_rssi", but the default parameter I read they are the same threshold.

I would like them to switch to closer parents if the RSSI is worse than -70dBm.

I cant understand from the description the difference between params.cnx_rssi, params.select_rssi and params.switch_rssi and I don't know how to set them to achieve my goal.

zhangyanjiaoesp, 2025-09-02:
@michaelsimp 
When an RSSI threshold is set but no suitable parent is found during scanning, the threshold will be gradually lowered until a connectable AP is discovered.

michaelsimp, 2025-09-02:
Originally I set the thresholds on receiving IP address event. Then I changed it to before starting wifi mesh. No change in behaviour.
I won’t be able to get you logs for a few days now as travelling back home.
I raised another ticket as I tried manual wifi mesh networking to get more visibility and control of the Wifi mesh. It would be great if you could respond on this as I think this might be a better option for me moving forwards