espressif/esp-idf #17499: [v5.1.4] Korg B2 Digital Piano hits `assert failed: _buffer_parse_bulk hcd_dwc.c:2271 (rem_len <= transfer->num_bytes)`

chipweinberger, 2025-08-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

N/A

### Power Supply used.

USB

### What is the expected behavior?

No crash

### What is the actual behavior?

Crash during USB transfer

```
==================== CURRENT THREAD STACK =====================
#0  0x40376056 in panic_abort (details=0x3fca6bb4 <port_IntStack+3828> "assert failed: _buffer_parse_bulk hcd_dwc.c:2271 (rem_len <= transfer->num_bytes)") at /esp-idf/components/esp_system/panic.c:466
#1  0x40384838 in esp_system_abort (details=0x3fca6bb4 <port_IntStack+3828> "assert failed: _buffer_parse_bulk hcd_dwc.c:2271 (rem_len <= transfer->num_bytes)") at /esp-idf/components/esp_system/port/esp_system_chip.c:84
#2  0x4038dcd4 in __assert_func (file=<optimized out>, line=<optimized out>, func=<optimized out>, expr=<optimized out>) at /esp-idf/components/newlib/assert.c:81
#3  0x4208d495 in _buffer_parse_bulk (buffer=0x3fcb9aec) at /esp-idf/components/usb/hcd_dwc.c:2271
#4  _buffer_parse (pipe=0x3c3e187c) at /esp-idf/components/usb/hcd_dwc.c:2406
#5  0x4208d898 in _intr_hdlr_chan (yield=<synthetic pointer>, chan_obj=<optimized out>, pipe=0x3c3e187c) at /esp-idf/components/usb/hcd_dwc.c:792
#6  intr_hdlr_main (arg=0x3fce8c74) at /esp-idf/components/usb/hcd_dwc.c:861
#7  0x40378688 in _xt_lowint1 () at /esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1240
#8  0x4038152d in xt_utils_wait_for_intr () at /esp-idf/components/xtensa/include/xt_utils.h:81
#9  esp_cpu_wait_for_intr () at /esp-idf/components/esp_hw_support/cpu.c:111
#10 0x42006ed5 in esp_vApplicationIdleHook () at /esp-idf/components/esp_system/freertos_hooks.c:59
#11 0x403871b0 in prvIdleTask (pvParameters=<optimized out>) at /esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4439
#12 0x403858a9 in vPortTaskWrapper (pxCode=0x403870fc <prvIdleTask>, pvParameters=0x0) at /esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162
```

```

================== CURRENT THREAD REGISTERS ===================
exccause       0x1d (StoreProhibitedCause)
excvaddr       0x0
epc1           0x42011ab6
epc2           0x0
epc3           0x0
epc4           0x0
epc5           0x0
epc6           0x0
eps2           0x0
eps3           0x0
eps4           0x0
eps5           0x0
eps6           0x0
pc             0x40376056          0x40376056 <panic_abort+34>
lbeg           0x40056f5c          1074098012
lend           0x40056f72          1074098034
lcount         0xffffffff          4294967295
sar            0x10                16
ps             0x60023             393251
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x80384838          -2143795144
a1             0x3fca6b10          1070230288
a2             0x3fca6bb4          1070230452
a3             0x2                 2
a4             0xa                 10
a5             0x3c3de490          1010689168
a6             0x3fcad7bc          1070258108
a7             0x3fcad7ac          1070258092
a8             0x0                 0
a9             0x1                 1
a10            0x3fca6ba3          1070230435
a11            0x3fca6ba3          1070230435
a12            0x3fca6ba4          1070230436
a13            0x3fcbc6e0          1070319328
a14            0x3fca92e0          1070240480
a15            0x3fca92d4          1070240468
```

### Steps to reproduce.

1. plug Korg B2 piano into USB port
2. play on the piano while the ESP32-S3 continually reads USB MIDI data

### Debug Logs.

```plain

```

chipweinberger, 2025-08-27:
@tore-espressif @roma-jam @peter-marcisovsky 

Thanks for your time. I know this is an older IDF at this point, but a targeted fix would still be appreciated if possible. Also perhaps it affects newer IDF as well.

chipweinberger, 2025-08-27:
Also, while I have you. Is the USB stack still under rapid development? I'd like to update but prefer to do it in a period of stability after a few rounds of bug fixes. This is a shipping product at this point.

tore-espressif, 2025-08-29:
Hi @chipweinberger thank you for the report.

As mentioned in another issue, we had troubles reproducing this bug. So now we can only offer a way how to silently ignore this error. A full USB device reset (essentially disconnection followed by connection) might be necessary.

However, we could get the same error by misusing the usb transfer API. Could you please confirm that you do not change `transfer->num_bytes` after you submitted the transfer? Or that you accidentaly zeroize the whole transfer (`usb_transfer_t`) somewhere while the transfer is ongoing? After you call `usb_host_transfer_submit()` you cannot alter the transfer's internal before the transfer finished (calls its callback)
Also, the transfer size must be max (128k - 1), this is not checked by the host lib :(

> Is the USB stack still under rapid development?

No, currently we have only few remaining (hard to fix) bugs. Most of the planned work now focuses on High Speed (relevant only for P4), adding support for upcoming targets (H4) -> no functional changes, and adding missing features (suspend/resume)