espressif/esp-idf #17500: DHCP client does not retry after decline with arp check

dorianim, 2025-08-27:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

When `CONFIG_LWIP_DHCP_DOES_ARP_CHECK` or `CONFIG_LWIP_DHCP_DOES_ACD_CHECK` is enabled and an address conflict is detected using the ARP/ACD check, the ESP should send DHCPDECLINE and retry after 10 seconds with a new DHCPDISCOVER.

### What is the actual behavior?

When `CONFIG_LWIP_DHCP_DOES_ARP_CHECK` is enabled and an address conflict is detected using the ARP check, the ESP sends DHCPDECLINE but **never retries**. **No new DHCPDISCOVER is sent, and the ESP never gets an IP.**

When `CONFIG_LWIP_DHCP_DOES_ACD_CHECK` is enabled, it seems to work just fine.

### Steps to reproduce.

1. Create a suitable scenario where an address conflict happens with the address offered to the ESP in DHCPOFFER.
2. The ESP sends DHCPDECLINE.
3. The ESP stalls (at this point it should try again with DHCPDISCOVER).

### Debug Logs.

```plain

```

### More Information.

As far as I understand, the lwIP function `dhcp_conflict_callback` expects to first get `ACD_DECLINE` to send the decline packet and then `ACD_RESTART_CLIENT` to restart the DHCP process:

https://github.com/espressif/esp-lwip/blob/fd432e4ee2cfb7f7f1c7eb7227e0173412e7b84e/src/core/ipv4/dhcp.c#L380-L401

When using `CONFIG_LWIP_DHCP_DOES_ACD_CHECK`, this seems to work as expected:
https://github.com/espressif/esp-lwip/blob/fd432e4ee2cfb7f7f1c7eb7227e0173412e7b84e/src/core/ipv4/acd.c

However, when using `CONFIG_LWIP_DHCP_DOES_ARP_CHECK`, it does not work, because `ACD_RESTART_CLIENT` is never sent:
https://github.com/espressif/esp-idf/blob/master/components/lwip/port/acd_dhcp_check.c

bmedici, 2026-03-10:
Any news on this one?
It's blocking on a project we're deploying for customers.

bmedici, 2026-04-08:
Hi, thank you for the fix and updated status. 

> * ✅ Included from: v6.0-beta1 and later (v6.0-beta2, v6.0-rc1, v6.0)
> * ❌ Not included in: v5.5.x (including v5.5.4)
> 
> If you're using v5.5.x, you can cherry-pick commit c2d87d962b4 into your branch.

I'm afraid this is not good news it's not at least fixed on the 5.5 branch, which is a pretty recent version and widely deployed. We even had hard times to upgrade to 5.x because of its memory footprint on ESP32. 

Cherry picking would imply derivating from general deployment and packages with local patches. 
What about making this fix (cherry picking the fix) on the official 5.x or 5.5.x branch on this official stream? 

Thank you!