espressif/esp-idf #17506: power management on wake up

donaldkohcw, 2025-08-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I would like to ask what is the best way to configure the ESP32-H2 to reduce power consumption if it only needs to wake up, perform a small calculation, and then return to deep sleep after 50 ms.

Currently, the current peaks at 15 mA for about 0.5 s after wake-up — I believe around 400 ms is due to the boot duration, and the remaining 100 ms is the execution time.

Do you have any advice on how to reduce the current draw or shorten the power-up time?

Regards,
Don

[image]

ESP-Marius, 2025-08-29:
Depending on what you need to do the [wake-up stub](https://docs.espressif.com/projects/esp-idf/en/stable/esp32h2/api-guides/deep-sleep-stub.html) (code ran straight away after wakeup, before boot process) might be an option

donaldkohcw, 2025-08-29:
Thanks, Marius. I would try. Is the 15mA expected during esp32h2 boot? anyway can reduce this?

closed 2026-08-27 by Alvin1Zhang