espressif/esp-idf #17515: Use of ADC_UNIT_2 in ULP coprocessor

deodatomatheus, 2025-09-01:
### Is your feature request related to a problem?

In the file **components/ulp/ulp_common/ulp_adc.c**, there is a check that triggers an error if the ADC is not ADC_UNIT_1:

`ESP_RETURN_ON_FALSE(cfg->adc_n == ADC_UNIT_1, ESP_ERR_INVALID_ARG, TAG, "Only ADC_UNIT_1 is supported for now");`

Is there a specific reason why other ADC units are not supported? I commented out this line and tested it on my board, and it worked fine using ADC_UNIT_2 CH3.

### Describe the solution you'd like.

I would like to be able to use ADC_UNIT_2.

### Describe alternatives you've considered.

Implement the necessary checks to allow using CH2 (for example, verifying potential conflicts with other drivers), or leave it up to the programmer.

SoucheSouche, 2025-09-04:
@deodatomatheus, as expected, the reason why the UNIT_2 was not supported in the first place is it is shared by other modules (WIFI/BT) and we cannot guaranty a safe usage from the low power core.

I could update adc_ulp.c to remove the limitation and leave the responsibility to the user to make sure that ADC2 is not used prior to initializing and using it from the lp core.

SoucheSouche, 2025-11-19:
@deodatomatheus,
I forgot to update you on the matter.
The use of UNIT_2 was added.

see 1e0463673b91d408bb7813a5e150c39c8cb71705