espressif/esp-idf #17521: (HFP_HF) esp_hf_client_audio_data_send() always returns error

walinsky, 2025-09-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I'm trying to integrate I2S in the [hfp_hf example](https://github.com/espressif/esp-idf/tree/v5.5.1/examples/bluetooth/bluedroid/classic_bt/hfp_hf) using the new (v5.5.x) API.
So far I've been successful in decoding the incoming MSBC packets using the [esp audio codec](https://components.espressif.com/components/espressif/esp_audio_codec/versions/2.3.0/readme), putting the PCM data on a ringbuffer, and send it to I2S.
Similarly I read mic data from I2S, MSBC encode it, and put it on a ringbuffer.
when bt_app_hf_client_audio_data_cb() is called, I send decoded SMBC to a ringbuffer, and I fetch SMBC encoded (mic) packets to send it to the lower layer (esp_hf_client_audio_data_send()).
esp_hf_client_audio_data_send() doens't fail; but immediately my esp32 reboots, reporting:
`assert failed: BTA_HfClientAudioDataSend bta_hf_client_api.c:397 (p_data - (UINT8 *)(p_buf + 1) >= 0)`

Although I'm pretty sure there's nothing wrong with my (57 byte SMBC encoded) data I'm trying to send, I've tried just copying the (incoming) SMBC data to a new esp_hf_audio_buff_t and send that to the lower layer.
Even that gives the same error:
`assert failed: BTA_HfClientAudioDataSend bta_hf_client_api.c:397 (p_data - (UINT8 *)(p_buf + 1) >= 0)`

This is the code I'm using for bt_app_hf_client_audio_data_cb()
`static void bt_app_hf_client_audio_data_cb(esp_hf_sync_conn_hdl_t sync_conn_hdl, esp_hf_audio_buff_t *audio_buf, bool is_bad_frame)
{
    if (!s_hfp_audio_connected) {
        esp_hf_client_audio_buff_free(audio_buf);
        return;
    }
    if (is_bad_frame) {
        esp_hf_client_audio_buff_free(audio_buf);
        return;
    }

    /* create a deep copy of audio_buf */
    esp_hf_audio_buff_t *audio_data_to_send = esp_hf_client_audio_buff_alloc((uint16_t) sizeof(esp_hf_audio_buff_t));
    uint8_t *mic_data = (uint8_t *) malloc (audio_buf->data_len);
    memcpy (mic_data, audio_buf->data, audio_buf->data_len);
    audio_data_to_send->buff_size = audio_buf->buff_size;
    audio_data_to_send->data_len = audio_buf->data_len;
    if (audio_data_to_send->data_len > ESP_HF_MSBC_ENCODED_FRAME_SIZE) {
        audio_data_to_send->data_len = ESP_HF_MSBC_ENCODED_FRAME_SIZE;
    }
    audio_data_to_send->data = mic_data;

    /* decode our incoming data and send it to i2s */
    esp_audio_dec_out_frame_t out_frame = {0};
    if (hfp_sbc_decoder(audio_buf->data, audio_buf->data_len, &out_frame)) {
        bt_i2s_hfp_write_tx_ringbuf(out_frame.buffer, out_frame.len);
        esp_hf_client_audio_buff_free(audio_buf);
    } else {
        esp_hf_client_audio_buff_free(audio_buf);
        return;
    }
    
    /* fetch our msbc encoded mic data and send it to the ag */
    // esp_hf_audio_buff_t *audio_data_to_send = esp_hf_client_audio_buff_alloc((uint16_t) sizeof(esp_hf_audio_buff_t));
    // uint8_t *mic_data = (uint8_t *) malloc (ESP_HF_MSBC_ENCODED_FRAME_SIZE);
    // size_t len = bt_i2s_hfp_read_rx_ringbuf(mic_data);
    // audio_data_to_send->buff_size = len;
    // audio_data_to_send->data_len = len;
    // audio_data_to_send->data = mic_data;

    if (s_msbc_air_mode && audio_data_to_send->data_len > ESP_HF_MSBC_ENCODED_FRAME_SIZE) {
        audio_data_to_send->data_len = ESP_HF_MSBC_ENCODED_FRAME_SIZE;
    }

    if (audio_data_to_send->data_len == 0 || audio_data_to_send->data == NULL) {
        esp_hf_client_audio_buff_free(audio_data_to_send);
        return;
    }

    if (esp_hf_client_audio_data_send(s_sync_conn_hdl, audio_data_to_send) != ESP_OK) {
        esp_hf_client_audio_buff_free(audio_data_to_send);
        ESP_LOGW(BT_HF_TAG, "%s failed to send audio data", __func__);
    }
}`

What's causing the error?
Is there something wrong with the way I'm creating the esp_hf_audio_buff_t ?

Tia

esp-lrh, 2025-09-09:
Hi, @walinsky 

The input parameter of `esp_hf_client_audio_buff_alloc` is the size of  encoded frame you want to send, for mSBC, you can use the constant value 57. 

correct usage:
```
    /* create a deep copy of audio_buf */
    esp_hf_audio_buff_t *audio_data_to_send = esp_hf_client_audio_buff_alloc(audio_buf->data_len);
    /* just copy data, data buff already allocated by esp_hf_client_audio_buff_alloc */
    memcpy (audio_data_to_send->data, audio_buf->data, audio_buf->data_len);
    // audio_data_to_send->buff_size = audio_buf->buff_size; // do not modify buff_size, here, it is equal to audio_buf->data_len
    audio_data_to_send->data_len = audio_buf->data_len;
    if (audio_data_to_send->data_len > ESP_HF_MSBC_ENCODED_FRAME_SIZE) {
        audio_data_to_send->data_len = ESP_HF_MSBC_ENCODED_FRAME_SIZE;
    }

    /* now we can send the audio buff */
    if (esp_hf_client_audio_data_send(s_sync_conn_hdl, audio_data_to_send) != ESP_OK) {
        esp_hf_client_audio_buff_free(audio_data_to_send);
        ESP_LOGW(BT_HF_TAG, "%s failed to send audio data", __func__);
    }
```

We can also forward or reuse the received audio buff, this can reduce one copy and free operation:
```
   if (audio_buf->buff_size >= ESP_HF_MSBC_ENCODED_FRAME_SIZE) {
        memcpy (audio_buf->data, encoded_msbc_data, ESP_HF_MSBC_ENCODED_FRAME_SIZE);
        audio_buf->data_len = ESP_HF_MSBC_ENCODED_FRAME_SIZE;
  
        if (esp_hf_client_audio_data_send(s_sync_conn_hdl, audio_buf) != ESP_OK) {
            esp_hf_client_audio_buff_free(audio_buf);
            ESP_LOGW(BT_HF_TAG, "%s failed to send audio data", __func__);
        }
    }
```

esp-lrh, 2025-09-09:
only noise or normal audio with noise?

`The decoder reports: frame sample rate: 16000, bits per sample: 16, channel(s): 1, bitrate: 60800, frame size: 17` 'frame size 17' doesn't look right, the frame size of hfp msbc should be 57

esp-lrh, 2025-09-16:
Hi, @walinsky 

It may not be a microphone/speakers problem. HFP uses SCO/eSCO for audio data transmission. For SCO, there is no retransmission, while for eSCO, there is limited retransmission. When the configuration option "BT_HFP_WBS ENABLE" is enabled, eSCO will be used first. 

Even though the connection uses a limited number of retransmissions, packet loss may still occur, so please calculate the packet loss rate during hfp voice calls, you can periodically call `esp_hf_client_pkt_stat_nums_get` to get the current statistics of sent and received packets (report in `ESP_HF_CLIENT_PKT_STAT_NUMS_GET_EVT`).

If there is a high packet loss rate, it may be that there is a problem with the ESP32 antenna, or there may be too many interference signals in the environment. Please update the results here and I will continue the analysis.

closed 2025-10-26 by walinsky