espressif/esp-idf #17528: rmt_transmit loop count is not supported

nopnop2002, 2025-09-04:
### Is your feature request related to a problem?

I used this sample.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/rmt/ir_nec_transceiver

I made the following changes:
```
    // this example won't send NEC frames in a loop
    rmt_transmit_config_t transmit_config = {
        //.loop_count = 0, // no loop
        .loop_count = 1, // no loop
    };
```

We got this error:
```
E (301) rmt: rmt_transmit(545): loop count is not supported
ESP_ERROR_CHECK failed: esp_err_t 0x106 (ESP_ERR_NOT_SUPPORTED) at 0x400d7149
--- 0x400d7149: app_main at /home/nop/rtos/ir_nec_transceiver/main/ir_nec_transceiver_main.c:240
file: "./main/ir_nec_transceiver_main.c" line 240
func: app_main
expression: rmt_transmit(tx_channel, nec_encoder, &scan_code_on, sizeof(scan_code_on), &transmit_config)

abort() was called at PC 0x400862ef on core 0
--- 0x400862ef: _esp_error_check_failed at /home/nop/esp-idf/components/esp_system/esp_err.c:49
```

### Describe the solution you'd like.

loop count is supported.

### Describe alternatives you've considered.

```
        const ir_nec_scan_code_t scan_code_on = {
            .address = 0x5284,
            .command = 0x956A,
        };
        for (int loop=0;loop<5;loop++) {
            ESP_ERROR_CHECK(rmt_transmit(tx_channel, nec_encoder, &scan_code_on, sizeof(scan_code_on), &transmit_config));
            vTaskDelay(10);
        }
```

nopnop2002, 2025-09-04:
I found this description.
```
the loop transmit feature is not supported on all ESP chips, please refer to [[TRM](https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf#rmt)] before you configure this option. Or you might encounter [ESP_ERR_NOT_SUPPORTED](https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32/api-reference/system/esp_err.html#c.ESP_ERR_NOT_SUPPORTED) error.
```

suda-morris, 2025-09-04:
Hi @nopnop2002 We've made an improvement recently: https://github.com/espressif/esp-idf/commit/f160701c29852e48411c776ad8e773f447ae1923