espressif/esp-idf #17529: Wifi Aware Nan console Crashes when trying to connect with Android

eswarm, 2025-09-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am running the WifiAware Nan Console sample application to test WifiAware. 
When I test it with two esp32 peers it is able to both publish and subscribe as expected. 

But when I try to publish or subscribe the service from a Android device it crashes with the below log. 

```

 (22780) wifi:Invalid time bitmap in Availabiliy
E (22780) wifi:Invalid time bitmap in Availabiliy
E (22780) wifi:Invalid time bitmap in Availabiliy
E (27790) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (27790) task_wdt:  - IDLE0 (CPU 0)
E (27790) task_wdt: Tasks currently running:
E (27790) task_wdt: CPU 0: wifi
E (27790) task_wdt: CPU 1: IDLE1
E (27790) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x400E0332:0x3FFB2040 0x400E0710:0x3FFB2060 0x40082D11:0x3FFB2090 0x4012AC51:0x3FFC06A0 0x401418CC:0x3FFC06C0 0x40141C79:0x3FFC06E0 0x4012C4A1:0x3FFC0700 0x4012C521:0x3FFC0750 0x4009302A:0x3FFC0770 0x4008FFA3:0x3FFC0790 0x40169236:0x3FFC07C0
--- 0x400e0332: task_wdt_timeout_handling at /Users/eswarm/esp/master/esp-idf/components/esp_system/task_wdt/task_wdt.c:436
--- 0x400e0710: task_wdt_isr at /Users/eswarm/esp/master/esp-idf/components/esp_system/task_wdt/task_wdt.c:509
--- 0x40082d11: _xt_lowint1 at /Users/eswarm/esp/master/esp-idf/components/xtensa/xtensa_vectors.S:1240
--- 0x4012ac51: nan_sched_parse_avail_entries at ??:?
--- 0x401418cc: nan_sd_parse_availability at ??:?
--- 0x40141c79: nan_rx_sdf at ??:?
--- 0x4012c4a1: nan_input at ??:?
--- 0x4012c521: nan_rx_cb at ??:?
--- 0x4009302a: ppRxPkt at ??:?
--- 0x4008ffa3: ppTask at ??:?
--- 0x40169236: vPortTaskWrapper at /Users/eswarm/esp/master/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139
```

Exact steps 

1. Start Nan discovery with   nan -S 
2. Publish a service with   publish -n TEST
3. Start the Android application, give the required permissions 
4. Start subscription on Android application. 
5. Observe the crash or sometimes it just print the log "wifi:Invalid time bitmap in Availabiliy"

eswarm, 2025-09-04:
The Wifi Aware Sample to reproduce on Android can be found here, https://github.com/eswarm/WifiAwareSample

YangJie09, 2025-11-03:
I have tow demos to test Wi-Fi Aware, iOS_demo and Android_demo , set the same service name: "_sat-simulation._udp" , but can not discover each other. what else could be going wrong?