espressif/esp-idf #17532: How to control SOC_MOD_CLK_PLL_D2

kaidegit, 2025-09-05:
### Is your feature request related to a problem?

I want to increase this because i2s clk is not high enough. Is there a way to control it and how high the clock can reach.

ginkgm, 2025-09-25:
Hi @kaidegit ,

May I know what's the chip you are using? And what clock frequency do you want?

Unfortunately, our PLL frequency cannot be changed. But maybe we can check if there are other clock source that can meet your requirement, for example, the configurable APLL.

kaidegit, 2025-09-25:
> Hi @kaidegit ,
> 
> May I know what's the chip you are using? And what clock frequency do you want?

esp32s3 and ads7046

ads7046 needs a 60m spi-like signal. spi seems cannot run in 60m. but i2s can simulate it.

I also see spi in c5 have a 120m clock, maybe spi in c5 can output a 60m signal?

> 
> Unfortunately, our PLL frequency cannot be changed. But maybe we can check if there are other clock source that can meet your requirement, for example, the configurable APLL.

I think d2 pll is apll? whatever, seems no api can configuate it.

L-KAYA, 2025-09-26:
@kaidegit D2 is not APLL, and APLL can only reach 125MHz.

I think I2S is not a proper peripheral to communicate with `ads7046`. Parlio RX and SPI should be the better choice.

> I also see spi in c5 have a 120m clock

SPI on C5 and C61 can use `SOC_MOD_CLK_PLL_F160M` as clock source, 60M clock should be OK on it.

kaidegit, 2025-09-26:
> D2 is not APLL, and APLL can only reach 125MHz.

So Which is APLL? I only see these in code

```c++
/**
 * @brief Array initializer for all supported clock sources of I2S
 */
#define SOC_I2S_CLKS {SOC_MOD_CLK_PLL_F240M, SOC_MOD_CLK_PLL_F160M, SOC_MOD_CLK_XTAL, I2S_CLK_SRC_EXTERNAL}

/**
 * @brief I2S clock source enum
 */
typedef enum {
    I2S_CLK_SRC_DEFAULT = SOC_MOD_CLK_PLL_F160M,                    /*!< Select PLL_F160M as the default source clock  */
    I2S_CLK_SRC_PLL_240M = SOC_MOD_CLK_PLL_D2,                      /*!< Select PLL_D2 as the source clock.
                                                                         It is default to 240MHz while PLL is 480MHz,
                                                                         but it will be 160MHz if PLL is 320MHz */
    I2S_CLK_SRC_PLL_160M = SOC_MOD_CLK_PLL_F160M,                   /*!< Select PLL_F160M as the source clock */
    I2S_CLK_SRC_XTAL = SOC_MOD_CLK_XTAL,                            /*!< Select XTAL as the source clock */
    I2S_CLK_SRC_EXTERNAL = -1,                                      /*!< Select external clock as source clock */
} soc_periph_i2s_clk_src_t;
```

wanckl, 2025-09-26:
@kaidegit Hi, not all clock is connect to SPI, and if you want an accurate 60M clock, for now only ESP32P4 is able to do it. other wise for C5 which has 160M source, it is able to generate 160/3=53M clock.

Then if you are using spi master driver in esp-idf, you can config speed in device_config structure instead of set clock divider by your self.