espressif/esp-idf #17533: Interrupt wdt timeout on CPU0 when calling  wifi_prov_mgr_reset_provisioning()

michaelsimp, 2025-09-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5.0

### Espressif SoC revision.

Chip is ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

I need to be able to unprovision my WiFi credentials (SSID and Password) programmatically.

I expect the wifi_prov_mgr_reset_provisioning() function to complete its task before I do a 5 second delay and call esp_restart().  After the reboot I expect the system NOT to be provisioned for Wifi.
I am using BLE Provisioning which is working fine.

### What is the actual behavior?

Calling wifi_prov_mgr_reset_provisioning() works a lot of the time but regularly fails

When it fails it reports,
` Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0), 
`
(see logs below)
After restart, the WiFi provisioning is still there.

### Steps to reproduce.

On request from my console task, the function wifiUnprovision() (see below) is called.

```
void espErrorCheck(bool showAll, esp_err_t err, const char* text) {
    if (err != ESP_OK) ESP_LOGE(TAG, "%s: %s", text, esp_err_to_name(err));
    else if (showAll) ESP_LOGI(TAG, "%s: %s", text, esp_err_to_name(err));
}
void wifiUnprovision(void) {
    ESP_LOGW(TAG, "Wifi provisioning reset");

    // prerequisites for unprovisioning
    wifi_prov_mgr_stop_provisioning();
    wifi_prov_mgr_config_t config = {
        .scheme = wifi_prov_scheme_ble, // Or wifi_prov_scheme_softap
        .scheme_event_handler = WIFI_PROV_SCHEME_BLE_EVENT_HANDLER_FREE_BTDM // Example event handler
    };
    espErrorCheck(true, wifi_prov_mgr_init(config),"wifi_prov_mgr_init");
    espErrorCheck(true, wifi_prov_mgr_reset_provisioning(),"wifi_prov_mgr_reset_provisioning");
    sleepMs(5000);
    esp_restart();
}
```

### Debug Logs.

```plain
W (13:13:58.945) aWifiProv: Wifi provisioning reset
E (13:13:58.946) wifi_prov_mgr: Provisioning manager not initialized
I (13:13:58.948) wifi_prov_scheme_ble: BT memory released
I (13:13:58.949) util: wifi_prov_mgr_init: ESP_OK
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4037eea9  PS      : 0x00060534  A0      : 0x803837f0  A1      : 0x3fcc7870
--- 0x4037eea9: esp_cpu_compare_and_set at C:/Users/micha/esp/v5.5/esp-idf/components/esp_hw_support/cpu.c:205

A2      : 0x3fce9f4c  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x3fccf544
A6      : 0x00000000  A7      : 0x3fcae290  A8      : 0x03ce9f4c  A9      : 0x01ffffff
A10     : 0x3fce9f4c  A11     : 0x3c291924  A12     : 0x00000002  A13     : 0x3fccf944
A14     : 0x4038a718  A15     : 0x00000000  SAR     : 0x00000013  EXCCAUSE: 0x00000005
--- 0x4038a718: esp_log_write at C:/Users/micha/esp/v5.5/esp-idf/components/log/src/os/log_write.c:39

EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00000000
--- 0x400570e8: memset in ROM
--- 0x400570f3: memset in ROM

Backtrace: 0x4037eea6:0x3fcc7870 0x403837ed:0x3fcc7890 0x403833c4:0x3fcc78c0 0x40377150:0x3fcc7900 0x40377239:0x3fcc7930 0x420d8f51:0x3fcc7950 0x420d60c1:0x3fcc7970 0x420d6130:0x3fcc79b0 0x421553fe:0x3fcc79d0 0x42155531:0x3fcc7a00 0x4215918c:0x3fcc7a20 0x42156562:0x3fcc7a40 0x4038c67d:0x3fcc7a60 0x403835e9:0x3fcc7a90
--- 0x4037eea6: esp_cpu_compare_and_set at C:/Users/micha/esp/v5.5/esp-idf/components/esp_hw_support/cpu.c:205
--- 0x403837ed: spinlock_acquire at C:/Users/micha/esp/v5.5/esp-idf/components/esp_hw_support/include/spinlock.h:132
---  (inlined by) xPortEnterCriticalTimeout at C:/Users/micha/esp/v5.5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:479
--- 0x403833c4: vPortEnterCritical at C:/Users/micha/esp/v5.5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:567
---  (inlined by) xQueueSemaphoreTake at C:/Users/micha/esp/v5.5/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1727
--- 0x40377150: lock_acquire_generic at C:/Users/micha/esp/v5.5/esp-idf/components/newlib/src/locks.c:160
--- 0x40377239: _lock_acquire at C:/Users/micha/esp/v5.5/esp-idf/components/newlib/src/locks.c:169
--- 0x420d8f51: nvs::Lock::Lock() at C:/Users/micha/esp/v5.5/esp-idf/components/nvs_flash/src/nvs_platform.cpp:23
--- 0x420d60c1: int nvs_set<unsigned char>(unsigned long, char const*, unsigned char) at C:/Users/micha/esp/v5.5/esp-idf/components/nvs_flash/src/nvs_api.cpp:360
--- 0x420d6130: nvs_set_u8 at C:/Users/micha/esp/v5.5/esp-idf/components/nvs_flash/src/nvs_api.cpp:378
--- 0x421553fe: wifi_nvs_validate_lr_info at ??:?
--- 0x42155531: wifi_nvs_restore at ??:?
--- 0x4215918c: wifi_restore_process at ??:?
--- 0x42156562: ieee80211_ioctl_process at ??:?
--- 0x4038c67d: ppTask at ??:?
--- 0x403835e9: vPortTaskWrapper at C:/Users/micha/esp/v5.5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

Core  1 register dump:
PC      : 0x4037efde  PS      : 0x00060534  A0      : 0x820052d1  A1      : 0x3fcbb3a0
--- 0x4037efde: esp_cpu_wait_for_intr at C:/Users/micha/esp/v5.5/esp-idf/components/esp_hw_support/cpu.c:64

A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x8038394d  A5      : 0x00060723
A6      : 0x00060023  A7      : 0x0000abab  A8      : 0x820d218a  A9      : 0x3fcbb360
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00060720  A13     : 0x3fcdcdb0
A14     : 0x00060023  A15     : 0x3fcbb570  SAR     : 0x0000001a  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x4037efdb:0x3fcbb3a0 0x420052ce:0x3fcbb3c0 0x40384b69:0x3fcbb3e0 0x403835e9:0x3fcbb400
--- 0x4037efdb: xt_utils_wait_for_intr at C:/Users/micha/esp/v5.5/esp-idf/components/xtensa/include/xt_utils.h:82
---  (inlined by) esp_cpu_wait_for_intr at C:/Users/micha/esp/v5.5/esp-idf/components/esp_hw_support/cpu.c:55
--- 0x420052ce: esp_vApplicationIdleHook at C:/Users/micha/esp/v5.5/esp-idf/components/esp_system/freertos_hooks.c:58
--- 0x40384b69: prvIdleTask at C:/Users/micha/esp/v5.5/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4350 (discriminator 1)
--- 0x403835e9: vPortTaskWrapper at C:/Users/micha/esp/v5.5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: c5b391030

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375f2d
etc...
```

### More Information.

I have not written any ISR routines.  I am using WiFiMesh and have tried stopping it as follows before calling my wifiUnprovision() function (above).

```
    mesh_netifs_stop();
    espErrorCheck(false, esp_event_handler_unregister(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler), "esp_event_handler_unregister");
    espErrorCheck(false, mesh_netifs_destroy(), "mesh_netifs_destroy");
    espErrorCheck(false, esp_mesh_stop(), "esp_mesh_stop");
    espErrorCheck(false, esp_mesh_deinit(), "esp_mesh_deinit");
 
    esp_wifi_stop();
    espErrorCheck(false, esp_event_handler_unregister(IP_EVENT, IP_EVENT_STA_GOT_IP, &ip_event_handler), "esp_event_handler_unregister");

```

michaelsimp, 2025-09-08:
Hi Alvin
Are you able to give me an estimate as to when a fix might be available. 
Ideally I would love a patch for IDF ver 5.5.0. Is this possible?

hrushikesh430, 2025-09-18:
Hello @Alvin1Zhang, I was trying to reproduce your issue but unable to reproduce. Could you please try on your side with just `wifiUnprovision` function to remove the credentials and restart on `esp-idf v5.5`, to check whether it gives same error. And if possible could you please provide me the `.patch` file your application with respective `sdkconfig` file. So that I can try out on my side. Thanks.