espressif/esp-idf #17534: ESP32+IP101+lwip在数据重发过程中，重发时间间隔有错误

Yan-ZhJ, 2025-09-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.5.1

### Espressif SoC revision.

ESP32(V3.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-Ethernet-Kit_A_V1.2

### Power Supply used.

USB

### What is the expected behavior?

使用Netconn API，测试数据和FIN重发情况，希望的重发时间间隔：0.5s、1s、2s、4s、8s、16s、32s、64s、64s、64s、64s、64s

### What is the actual behavior?

重启后第一次的重发间隔时间正常，重发超时断开连接后，再次连接时的重发时间间隔发生错误，时间间隔出现2s、4s、8s、 ......；或者3s、6s、12s、......；还有其他的时间间隔情况。

### Steps to reproduce.

自定义的menuconfig
```
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_PARTITION_TABLE_SINGLE_APP_LARGE=y
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
CONFIG_COMPILER_OPTIMIZATION_PERF=y
CONFIG_FREERTOS_HZ=1000
CONFIG_SPIRAM=y
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_LWIP_ENABLE=y
CONFIG_LWIP_LOCAL_HOSTNAME="DATA-LINK"
CONFIG_LWIP_TCP_MSS=1460
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=11680
CONFIG_LWIP_TCP_WND_DEFAULT=11680
CONFIG_LWIP_TCP_RECVMBOX_SIZE=10
# CONFIG_LWIP_TCP_HIGH_SPEED_RETRANSMISSION is not set
CONFIG_LWIP_TCP_RTO_TIME=5000
CONFIG_LWIP_UDP_RECVMBOX_SIZE=10
CONFIG_LWIP_TCP_SYNMAXRTX=4
CONFIG_LWIP_TCP_MSL=600000
CONFIG_LWIP_TCP_FIN_WAIT_TIMEOUT=600000
CONFIG_LWIP_TCP_MAXRTX=10
# CONFIG_LWIP_IPV6 is not set
```

TCP连接任务
[... 181 lines cut ...]
            xSemaphoreGive(xMutex);
        }
        rsnum = 0;
        if (ethernet_link) {
            // 显示CONNECTION TIMEOUT，这个条件是重发数据失败
            // sprintf(sbuf, MSG_CLOSE, p_prog->command);
            sprintf(sbuf, MSG_CONNECT_TIMEOUT, p_prog->command);
        } else {
            sprintf(sbuf, MSG_CONNECT_RST, p_prog->command);
        }
        cmd_print(sbuf, RS_COM, p_prog->rmsg);
        set_connect_stat(p_prog, 0);
        break;
}

···

连接成功后断开对手设备网线，发送数据，观察wireshark数据重发情况，重启后第一次连接的重发时间间隔正常。
超时断开连接后，再次进行TCP连接，连接成功后再次断开对手设备网线，发送数据，观察wireshark数据重发情况，这次的数据重发时间间隔就会出现问题。

### Debug Logs.

```plain

```

Yan-ZhJ, 2025-09-08:
补充
设备重启后，第一次进行TCP连接时，如果对手设备不存在，发生SYN重发，重发过程中设备对手设备连接上，TCP建立成功后，进行数据发送，此时如果出现数据重发，数据重发的间隔时间也是错误的。

Yan-ZhJ, 2025-09-08:
RTO 参数明显出错
D (51443) lwip: tcp_receive: RTO 1 (500 milliseconds)
D (150271) lwip: tcp_receive: RTO 4 (2000 milliseconds)

LOG 记录
I (70) HOOKS: This hook is called BEFORE bootloader initialization
I (33) boot: ESP-IDF v5.5.1-dirty 2nd stage bootloader
I (33) boot: compile time Sep  8 2025 09:27:36
I (33) boot: Multicore bootloader
I (35) boot: chip revision: v3.1
I (38) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (45) boot.esp32: SPI Flash Size : 8MB
I (49) boot: Enabling RNG early entropy source...
I (53) HOOKS: This hook is called AFTER bootloader initialization
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (81) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 factory          factory app      00 00 00010000 00100000
I (94) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (101) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (107) boot: End of partition table
I (111) boot: Defaulting to factory image
I (114) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1e37ch (123772) map
I (153) esp_image: segment 1: paddr=0002e3a4 vaddr=3ffb0000 size=01c74h (  7284) load
I (155) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=53ad8h (342744) map
I (242) esp_image: segment 3: paddr=00083b00 vaddr=3ffb1c74 size=00c94h (  3220) load
I (244) esp_image: segment 4: paddr=0008479c vaddr=40080000 size=12300h ( 74496) load
I (268) esp_image: segment 5: paddr=00096aa4 vaddr=50000000 size=00020h (    32) load
I (277) boot: Loaded app from partition at offset 0x10000
I (277) boot: Disabling RNG early entropy source...
I (288) quad_psram: This chip is ESP32-D0WD
I (288) esp_psram: Found 8MB PSRAM device
I (289) esp_psram: Speed: 40MHz
I (289) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (296) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (304) cpu_start: Multicore app
D (307) cpu_start: Pro cpu up
D (309) cpu_start: Starting app cpu, entry point is 0x400814d0
--- 0x400814d0: call_start_cpu1 at C:/Users/Yan/esp/v5.5.1/esp-idf/components/esp_system/port/cpu_start.c:197
D (0) cpu_start: App cpu up
I (1172) esp_psram: SPI SRAM memory test OK
D (1180) clk: RTC_SLOW_CLK calibration value: 3056474
D (1185) cpu_start: calling init function: 0x400d34c8 on core: 0
--- 0x400d34c8: __esp_system_init_fn_init_efuse_check at C:/Users/Yan/esp/v5.5.1/esp-idf/components/efuse/src/esp_efuse_startup.c:37
D (1185) cpu_start: calling init function: 0x400d3984 on core: 0
--- 0x400d3984: __esp_system_init_fn_init_show_cpu_freq at C:/Users/Yan/esp/v5.5.1/esp-idf/components/esp_system/startup_funcs.c:64
I (1187) cpu_start: Pro cpu start user code
I (1191) cpu_start: cpu freq: 160000000 Hz
D (1195) cpu_start: calling init function: 0x400d3400 on core: 0
--- 0x400d3400: __esp_system_init_fn_init_show_app_info at C:/Users/Yan/esp/v5.5.1/esp-idf/components/esp_app_format/esp_app_desc.c:123
I (1200) app_init: Application information:
I (1204) app_init: Project name:     PoE-T9
I (1208) app_init: App version:      1
I (1212) app_init: Compile time:     Sep  8 2025 11:38:00
I (1217) app_init: ELF file SHA256:  4bb9f1a2c...
I (1221) app_init: ESP-IDF:          v5.5.1-dirty
D (1226) cpu_start: calling init function: 0x400d34d8 on core: 0
--- 0x400d34d8: __esp_system_init_fn_init_efuse_show_app_info at C:/Users/Yan/esp/v5.5.1/esp-idf/components/efuse/src/esp_efuse_startup.c:47
I (1231) efuse_init: Min chip rev:     v0.0
I (1235) efuse_init: Max chip rev:     v3.99
I (1239) efuse_init: Chip rev:         v3.1
D (1243) cpu_start: calling init function: 0x400d5248 on core: 0
--- 0x400d5248: __esp_system_init_fn_init_heap at C:/Users/Yan/esp/v5.5.1/esp-idf/components/heap/heap_caps_init.c:26
D (1249) memory_layout: Checking 10 reserved memory ranges:
D (1254) memory_layout: Reserved memory range 0x3f800000 - 0x3fc00000
D (1260) memory_layout: Reserved memory range 0x3ff82000 - 0x3ff82000
D (1267) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
[... 427 lines cut ...]
--- 0x400dbf3c: got_ip_event_handler at E:/Workspaces/vscode/Enet-T25/main/main.c:271
D (189322) lwip: tcp_slowtmr: rtime 32 pcb->rto 32

D (189322) lwip: tcp_output_segment: rtseq 2794216846

D (189322) event: running post IP_EVENT:8 with handler 0x400dbf3c and context 0x3ffc1f4c on loop 0x3ffb6cd4
--- 0x400dbf3c: got_ip_event_handler at E:/Workspaces/vscode/Enet-T25/main/main.c:271
I (189334) tcp_receive_task: WAITING 3 rsnum:4, nrtx:4
D (221323) lwip: tcp_slowtmr: rtime 64 pcb->rto 64

D (221323) lwip: tcp_output_segment: rtseq 2794216846

D (221323) event: running post IP_EVENT:8 with handler 0x400dbf3c and context 0x3ffc1f4c on loop 0x3ffb6cd4
--- 0x400dbf3c: got_ip_event_handler at E:/Workspaces/vscode/Enet-T25/main/main.c:271
I (221335) tcp_receive_task: WAITING 3 rsnum:5, nrtx:5
D (236961) esp.emac: receive len= 60
D (236962) event: running post IP_EVENT:8 with handler 0x400dbf3c and context 0x3ffc1f4c on loop 0x3ffb6cd4
--- 0x400dbf3c: got_ip_event_handler at E:/Workspaces/vscode/Enet-T25/main/main.c:271
D (245037) event: running post IP_EVENT:8 with handler 0x400dbf3c and context 0x3ffc1f4c on loop 0x3ffb6cd4
--- 0x400dbf3c: got_ip_event_handler at E:/Workspaces/vscode/Enet-T25/main/main.c:271
D (245038) esp.emac: receive len= 60
D (245039) event: running post IP_EVENT:8 with handler 0x400dbf3c and context 0x3ffc1f4c on loop 0x3ffb6cd4
--- 0x400dbf3c: got_ip_event_handler at E:/Workspaces/vscode/Enet-T25/main/main.c:271
D (245049) event: running post IP_EVENT:8 with handler 0x400dbf3c and context 0x3ffc1f4c on loop 0x3ffb6cd4
--- 0x400dbf3c: got_ip_event_handler at E:/Workspaces/vscode/Enet-T25/main/main.c:271

david-cermak, 2025-10-01:
Please do not mix TCP API (`tcp_abort()`) and netconn API, this is not supported. Use `netconn_close()` and `netconn_delete()` only.

About RTO:
* lwIP recomputes the base RTO after the first successful ACK using the SRTT/RTTVAR estimator; from then on, retransmissions use that updated base with exponential backoff. This often lands at 3 slow-timer ticks (~1.5 s), producing 1.5s, 3s, 6s… sequences on later connections. This is expected given the code paths.
* Initial RTO is set from CONFIG_LWIP_TCP_RTO_TIME and TCP_SLOW_INTERVAL (500 ms). After any successful data/ACK, pcb->rto = (sa>>3)+sv replaces the initial RTO.
* The backoff table uses exponents with a left shift, so the pattern is base, 2×base, 4×base, ...