espressif/esp-idf #17543: Overzealous logging in I2C, add an option to turn it off

eriksl, 2025-09-07:
### Is your feature request related to a problem?

This is especially targeted toward the "i2c_master_execute_defined_operations" function. I assume this function is supplied allowing non-standard devices to be usable from IDF. Well I think it's a bit, let's say "contra-intuitive" that the IDF code bloats (my limited) log space with reports of "errors" that are to be expected. I am losing interesting log messages due to this.

So please, either add an option to NOT LOG but just return an error code to the calling code (which I think, is quite a common and expected way to do things) OR remove all logging from the i2c_master_execute_defined_operations function, as some "errors" are to be expected anyway. The i2c_master_probe function already does this, apparently (which is logical).

jrahlf, 2025-09-09:
@mythbuster5 please push it to release/5.5 branch as well

mythbuster5, 2025-09-09:
@jrahlf Hi, In version v6.0, we remove nack log and use a unique error code stands for nack. This we cannot backport to 5.5 becuase this will introduce  a breaking change. I understand your requirement. If it's urgently used in 5.5, is it possible for you to bypass this in your code directly?

eriksl, 2025-09-09:
I worked around it in git clone. This is what I had to do get rid of ALL logging on "transaction failures" that are to be expected when doing user defined transactions (e.g. probing/detecting of devices). Again I don't think this function should log anything at all, the caller is supposed to know what they're doing, even if it's shooting in the foot...

This just workaround changes, it's not pretty.
```
diff --git a/components/esp_driver_i2c/i2c_master.c b/components/esp_driver_i2c/i2c_master.c
index 562e6788a3..21efd585fb 100644
--- a/components/esp_driver_i2c/i2c_master.c
+++ b/components/esp_driver_i2c/i2c_master.c
@@ -155,11 +155,11 @@ static void s_i2c_err_log_print(i2c_master_event_t event, bool bypass_nack_log)
     if (event == I2C_EVENT_TIMEOUT) {
         ESP_LOGE(TAG, "I2C transaction timeout detected");
     }
-    if (bypass_nack_log != true) {
-        if (event == I2C_EVENT_NACK) {
-            ESP_LOGE(TAG, "I2C transaction unexpected nack detected");
-        }
-    }
+//    if (bypass_nack_log != true) {
+//        if (event == I2C_EVENT_NACK) {
+//            ESP_LOGE(TAG, "I2C transaction unexpected nack detected");
+//        }
+//    }
 }
 
 //////////////////////////////////////I2C operation functions////////////////////////////////////////////
@@ -942,13 +942,10 @@ static esp_err_t s_i2c_synchronous_transaction(i2c_master_dev_handle_t i2c_dev,
     i2c_dev->master_bus->trans_finish = false;
     i2c_dev->master_bus->queue_trans = false;
     i2c_dev->master_bus->ack_check_disable = i2c_dev->ack_check_disable;
-    ESP_GOTO_ON_ERROR(s_i2c_transaction_start(i2c_dev, timeout_ms), err, TAG, "I2C transaction failed");
-    xSemaphoreGive(i2c_dev->master_bus->bus_lock_mux);
-    return ret;
+    //ESP_GOTO_ON_ERROR(s_i2c_transaction_start(i2c_dev, timeout_ms), err, TAG, "I2C transaction failed");
+	if((ret = s_i2c_transaction_start(i2c_dev, timeout_ms) != ESP_OK))
+    	s_i2c_hw_fsm_reset(i2c_dev->master_bus, false);
 
-err:
-    // When error occurs, reset hardware fsm in case not influence following transactions.
-    s_i2c_hw_fsm_reset(i2c_dev->master_bus, false);
     xSemaphoreGive(i2c_dev->master_bus->bus_lock_mux);
     return ret;
 }
@@ -1397,12 +1394,14 @@ esp_err_t i2c_master_execute_defined_operations(i2c_master_dev_handle_t i2c_dev,
         }
     }
 
-    if (i2c_dev->master_bus->async_trans == false) {
-        ESP_RETURN_ON_ERROR(s_i2c_synchronous_transaction(i2c_dev, i2c_ops, operation_list_num, xfer_timeout_ms), TAG, "I2C transaction failed");
-    } else {
-        ESP_RETURN_ON_ERROR(s_i2c_asynchronous_transaction(i2c_dev, i2c_ops, operation_list_num, xfer_timeout_ms), TAG, "I2C transaction failed");
-    }
-    return ESP_OK;
+	esp_err_t esp_err;
+
+    if (i2c_dev->master_bus->async_trans == false)
+        esp_err = (s_i2c_synchronous_transaction(i2c_dev, i2c_ops, operation_list_num, xfer_timeout_ms));
+    else
+        esp_err = (s_i2c_asynchronous_transaction(i2c_dev, i2c_ops, operation_list_num, xfer_timeout_ms));
+
+    return(esp_err);
 }
  esp_err_t i2c_master_register_event_callbacks(i2c_master_dev_handle_t i2c_dev, const i2c_master_event_callbacks_t *cbs, void *user_data)
```

mythbuster5, 2025-09-15:
> Without breaking the API an additional function could be introduce to set logging level specifically for the I2C driver code. I don't think IDF code should be logging without explicit consent anyway. Errors should be passed as error codes and the user application should handle the exception (and maybe log the exception).

Yes, that's is what we did in master branch, and we will backport it to older branches, without change error code.

robert-hh, 2025-09-15:
> Yes, that's is what we did in master branch, and we will backport it to older branches,

@mythbuster5 Do you know a timeline for this backport?

jrahlf, 2025-09-19:
Um what about the "pull up resistor warning": https://github.com/espressif/esp-idf/blob/166269fb9338607aa9726ecc4ea2d1763de31f0e/components/esp_driver_i2c/i2c_master.c#L1041
This is still there on master branch and should also be removed in my opinion.

mythbuster5, 2025-09-19:
@jrahlf  Removing now

closed 2025-10-16 by mythbuster5