espressif/esp-idf #17552: 【SPI 从机半双工通信】ESP32-S3 无法直接接收外设摄像头（BF20A6）数据，如何跳过指令校验？

TooUpper, 2025-09-09:
我正在使用 ESP32-S3 通过 SPI 从机半双工模式连接外设摄像头模组 BF20A6 进行数据采集，但在测试过程中遇到问题：ESP32 无法接收到数据。通过逻辑分析仪确认，摄像头模组确实有数据输出。

初步排查发现，ESP32 的 SPI 从机半双工通信需先接收指令，再根据指令收发数据，而该摄像头模组会直接持续输出数据，无法插入指令帧。因此想请教：是否有方法可以关闭或跳过 ESP32 SPI 从机的指令校验机制，使其能够直接接收数据？

此外，在模组提供的驱动代码中使用了 spi_slave_hd_slot_config_t 结构体中的 ms_data_bitlen 字段，但我们尝试多个版本的 ESP-IDF 均未找到该字段，不确定是否为版本兼容性问题。如有相关经验或建议，恳请指点。

非常感谢！

补充信息：

MCU：ESP32-S3

外设模组：BF20A6

通信模式：SPI 从机半双工

尝试过的操作：

使用逻辑分析仪确认摄像头有数据输出；

检查多个 ESP-IDF 版本（未列出具体版本可补充），均未找到 ms_data_bitlen 字段。

期待您的帮助！

wanckl, 2025-09-09:
@TooUpper 摄像头是单线还是多线数据输出？如果是单线，可以使用 slave 驱动只配置单方向的数据传输，不要使用slave_hd驱动。很多摄像头都是这样接收的。不过目前驱动模式可能丢数，我们有修改版本的驱动可以使用。

至于 `spi_slave_hd_slot_config_t::ms_data_bitlen`，没找到这个组合，你是不是看错了，在哪里看到的？

TooUpper, 2025-09-09:
目前采用单线模式，尝试过标准 SPI从机模式，但是存在一个问题：该模式依赖 CS片选信号标识事务的起始，但 BF20A6 并不支持对 CS 信号的控制功能。若将SPI配置为从机模式，CS引脚会被固定为输入模式，无法通过软件主动拉低。也就没法控制接收。
关于spi_slave_hd_slot_config_t::ms_data_bitlen这一字段，我是在在官方提供的 BF20A6 驱动中看到的，但我们在IDF框架中进行了多次排查，也确实找不到，官方还没发给验证，很烦。

 kay
***@***.***
19145466784
Blog: www.tooupper.com

Original:

TooUpper, 2025-09-18:
> Hi [@TooUpper](https://github.com/TooUpper) 可以参考这个修改驱动的 [SPI Camera](https://github.com/PumpkinLin314/spi_camera) 实现，并自行添加 BF20A6 摄像头的配置。 注意事项和配置限制请参考工程中的 README 文档。

这个模块因为成本及其他一些客观原因，公司最终决定放弃了。非常感谢大家的回复和指导！