espressif/esp-idf #17553: Error Installing ESP-IDF VS Code Extension

VicRoss, 2025-09-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect the installation to complete after pressing Install

### What is the actual behavior?

I receive the following error:

"G:\esp\tools\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)

### Steps to reproduce.

1.  Add ESP IDF extension to VS Code
2.  Run the command to configure ESP-IDF
3.  Enter required fields

[image]

4.  Press the Install button
5. Install fails with error ...

[image]

### Build or installation Logs.

```plain
Warning File G:\esp\tools\tools\idf-git\2.39.2\usr\ssl\openssl.cnf
                                        already exists and was not updated.
Warning File G:\esp\tools\tools\idf-git\2.39.2\usr\ssl\openssl.cnf.dist
                                        already exists and was not updated.
Extracted G:\esp\tools\tools\idf-git\2.39.2 ...
Using existing G:\esp\tools\tools\idf-python\3.11.2
"G:\esp\tools\tools\idf-python\3.11.2\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)
```

### Diagnostic report archive.

[idf-diag-2d3779f6-6d0e-46d3-8534-f86cdba2aedb.zip](https://github.com/user-attachments/files/22225970/idf-diag-2d3779f6-6d0e-46d3-8534-f86cdba2aedb.zip)

### More Information.

I found a previous issue which appears to be the same, but there is no resolution provided.

(ERROR_INVALID_PYTHON) (IDFGH-14896) #15610

brianignacio5, 2025-09-09:
Please rename `G:\esp\tools\tools\idf-python\3.11.2` and run the setup again.

The extension download a copy of idf-python that includes pip but for some reason the one you previously have doesn't include it.

brianignacio5, 2025-09-10:
G:\esp\tools\python_env\idf5.5_py3.11_env\Scripts is the virtual env created with G:\esp\tools\tools\idf-python\3.11.2. If you installed ESP-IDF with the IDF Installer or other way, you can use the `Find ESP-IDF in your system` option from `Select ESp-IDF Version` dropdown in the setup wizard to re-use an existing esp-idf setup.