espressif/esp-idf #17555: How to create a <START><address+READ><STOP> (zero bytes to read) sequence in i2c_master_execute_defined_operations

eriksl, 2025-09-09:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This works as expected for a write sequence, i.e.

<START><address+WRITE><STOP>

This is processed as expected.

But if I change the first byte <address+WRITE> to <address+READ>, the STOP command is ignored and never sent. If try to workaround this and add an "empty" READ command (zero bytes with NACK enabled), 256 bytes are read. Off by one error?

This is not a major issue, but for probing devices, it's good to not only have a "write" probe (which I am using now) and a "read" probe (which may be required for some exotic devices).

Hopefully easy to fix.

mythbuster5, 2025-09-10:
It works fine on my side.

[image]

[image]

mythbuster5, 2025-09-15:
@eriksl Have you checked this topic?

eriksl, 2025-09-15:
Ok my findings. I did hope I'd find a flaw on my side, but apparently not yet.

This is the code, which, when called with receive_buffer == 0 to the same code as yours, with one small difference, when reading 0 bytes, I also pass a NULL ptr. Is that illegal here?

```
	address = (uint8_t)((address_in << i2c_address_shift) | i2c_read_flag);

	current = 0;

	i2c_operations[current++].command = I2C_MASTER_CMD_START;

	i2c_operations[current].command = I2C_MASTER_CMD_WRITE;
	i2c_operations[current].write.ack_check = true;
	i2c_operations[current].write.data = &address;
	i2c_operations[current].write.total_bytes = 1;
	current++;

	if(receive_buffer_size > 1)
	{
		i2c_operations[current].command = I2C_MASTER_CMD_READ;
		i2c_operations[current].read.ack_value = I2C_ACK_VAL;
		i2c_operations[current].read.data = receive_buffer;
		i2c_operations[current].read.total_bytes = receive_buffer_size - 1;
		current++;
	}

	i2c_operations[current].command = I2C_MASTER_CMD_READ;
	i2c_operations[current].read.ack_value = I2C_NACK_VAL;
	i2c_operations[current].read.data = receive_buffer ? &receive_buffer[receive_buffer_size - 1] : nullptr;
	i2c_operations[current].read.total_bytes = receive_buffer ? 1 : 0;
	current++;

	i2c_operations[current++].command = I2C_MASTER_CMD_STOP;

	rv = i2c_master_execute_defined_operations(data->device_handle, i2c_operations, current, 500);
```
The device 0x70 I am probing is still the good 'ole bus mux that accepts writes and reads of an arbitrary amount of bytes. I think you are using a nonexisting device (address) which cause the address to be NAKed already. This fine for devices not present, but what's happening when the device is present, is a tad ugly, see below. First picture is zoomed in the start of the transaction, see the address gets an ACK here. Then "a number" of bytes is read and NAKed, the slave device disconnects (hence the 0xff) but the ESP keeps reading. It's followed by a zoomed version of the whole transaction.

[image]

[image]

It's not shown here, but the transaction is finished with a STOP condition. Interesting how the reads actually get ACKed from a certain point. To me this looks like a decrement-before-test so 0x00 becomes 0xff before it's tested.

mythbuster5, 2025-09-16:
OK, I understand.

You can use following code to bypass this
```
	address = (uint8_t)((address_in << i2c_address_shift) | i2c_read_flag);

	current = 0;

	i2c_operations[current++].command = I2C_MASTER_CMD_START;

	i2c_operations[current].command = I2C_MASTER_CMD_WRITE;
	i2c_operations[current].write.ack_check = true;
	i2c_operations[current].write.data = &address;
	i2c_operations[current].write.total_bytes = 1;
	current++;

	if(receive_buffer_size > 1)
	{
		i2c_operations[current].command = I2C_MASTER_CMD_READ;
		i2c_operations[current].read.ack_value = I2C_ACK_VAL;
		i2c_operations[current].read.data = receive_buffer;
		i2c_operations[current].read.total_bytes = receive_buffer_size - 1;
		current++;
	}

        if(receive_buffer_size != 0)
        {
	i2c_operations[current].command = I2C_MASTER_CMD_READ;
	i2c_operations[current].read.ack_value = I2C_NACK_VAL;
	i2c_operations[current].read.data = receive_buffer ? &receive_buffer[receive_buffer_size - 1] : nullptr;
	i2c_operations[current].read.total_bytes = receive_buffer ? 1 : 0;
	current++;
        } 

	i2c_operations[current++].command = I2C_MASTER_CMD_STOP;

	rv = i2c_master_execute_defined_operations(data->device_handle, i2c_operations, current, 500);

```

We will try to fugure out how to properly fix it.

eriksl, 2025-09-16:
As in: I've got the fix ;-)

Seriously, thanks for looking into it.

HaneleBersy, 2025-09-19:
Hi,

This is an interesting question.

I don't know on which target you are working, but I've looked in the ESP32-S3 reference manual on the I2C chapter [https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf]). On pages 993-994, the chapter "Command Configuration" specifies that the length of data to be read or written (byte_num) must be between 1 and 255...

eriksl, 2025-09-19:
Surely interesting, if only because a write with 0 bytes written is possible and working. So I'd expect a read with zero bytes to work too. I think the limitation counts when you really send a READ command. But it may very well work without a READ command at all, just sending the ADDRESS plus READ bit. The problem is that when you do this, no STOP is sent (even though the sequence is terminated by a STOP/END command. Otherwise it would be just fine.

HaneleBersy, 2025-09-19:
I see that in the API there are only those commands:
`    I2C_MASTER_CMD_START,    /**< Start or Restart condition */
    I2C_MASTER_CMD_WRITE,    /**< Write operation */
    I2C_MASTER_CMD_READ,     /**< Read operation */
    I2C_MASTER_CMD_STOP,     /**< Stop condition */`

However in the ref manual, there is also an END command (opcode = 4). Maybe you could try it? Just by reading I don't get the nuance to the STOP command.

HaneleBersy, 2025-09-19:
> Surely interesting, if only because a write with 0 bytes written is possible and working. So I'd expect a read with zero bytes to work too. I think the limitation counts when you really send a READ [...]

Except that in your write with 0 bytes I suppose you do one WRITE command to send ADDRESS|W ( =1 byte) with ACK enabled. And then you do directly a STOP command. So there is no WRITE with 0 bytes effectively taking place and the I2C controller is in no weird state.

In contrary with the read with 0 bytes. Initially you wanted to push the I2C controller to its READ state through the extra READ command but with zero bytes to read, which is in contradiction with the ref manual. That's why @mythbuster5's solution makes sense. 

SDA remaining low, I think, means the master is still holding onto it, bc else there would be some clock activity.

What is the esp_err_t output you get from the i2c_master_execute_defined_operations() ?