espressif/esp-idf #17584: Having i2c_master_execute_defined_operations atomically/synchronuously

eriksl, 2025-09-14:
### Is your feature request related to a problem?

On ESP32-S3, when I construct an I2C transaction like this, using i2c_master_execute_defined_operations: <START><address+WRITE><data><data><STOP> (or <RESTART>) I can do it in two ways.

One is START, WRITE, STOP, where the WRITE command send the address and the data. The other one is like START, WRITE, WRITE, STOP where the address and data are sent separately in two WRITE commands. It works both ways, but the second method introduces a bus idle gap. Just like the hardware is waiting for the next command to be handed. So I am using the first method, but it's a bit more work, involves copying an array just to prepend the address.

[image]

### Describe the solution you'd like.

Maybe it's possible to fill the hardware command queue beforehand? Or fix this another way?

mythbuster5, 2025-09-15:
Will check. But this part will appear in task-isr switch, altough it's look like a little bit strange, but it should'n be an issue in I2C protocal.

BTW, I'm sorry but if you have more issue about `i2c_master_execute_defined_operations` interface, you can list them in one github ticktet, thanks!

mythbuster5, 2025-09-15:
Each write in i2c_master_execute_defined_operations will be treated as a relatively independent action. This allows user parse many write operations. In this case, we don't simply merge write operations, this will produce many situations (like handle both command list and fifo space) and make software complicated to consider all.

If user minds the idle gap, who can merge write command in one write operation.

Considered that this behavior follows I2C protocal. We will not adjust this behavior in near future.