espressif/esp-idf #17587: Wi-Fi Mesh Network Stability Issues with ESP32-C6

mightChamp, 2025-09-15:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

Site Scenario:

1) There are 10 Wi-Fi Mesh Networks (50 devices per network, total 500 devices).
2) The setup is along a two-lane road.
3) Each lane contains 5 mesh networks such as, 2 networks positioned opposite to each other across the road.
4) In each network, there are 50 devices placed in a line, 30 meters apart.
5) All networks are configured to use channel 0, but with different Network IDs and Passwords.
6) Wi-Fi is operating in LR Mode.
7) Max 10 devices can join a node, and 25 Max layer
8) At site max layer is 6
9) RRSI range -60 to -90
10) Hardware: ESP32-C6-WROOM-1, running ESP-IDF v5.3.4.

Issues Observed:

1) Most devices frequently connect and disconnect within 1–2 minutes and switch parent. Testing with different hardware produced the same results.
2) Some random devices fail to connect to the network entirely and remain disconnected most of the time.
3) , and), can cause

Request for Support:

Could you please help us diagnose these issues? Specifically:

What are the best practices to debug mesh network stability problems?
Are there recommended sniffing/monitoring tools or methods to identify the root cause?

Your guidance would be highly valuable.

zhangyanjiaoesp, 2025-09-15:
@mightChamp 

1. > In each network, there are 50 devices placed in a line, 30 meters apart.

Is the mesh network in tree topology or line topology? Is the distance between every two devices 30 meters? Since you arranged the devices in a line, a linear topology might be better.

2. > All networks are configured to use channel 0, but with different Network IDs and Passwords.

Since you have already set different mesh IDs and passwords for each network, they cannot communicate with each other. Why not assign them to different channels? If all networks are set to channel 0, they will default to channel 1, which means 500 devices will be communicating on the same channel — that’s extremely congested.

3. > Wi-Fi is operating in LR Mode.

Why do you insist on using LR mode? Especially if the distance between every two nodes is 30 meters.

4. > What are the best practices to debug mesh network stability problems?

There’s no simple method; it’s mainly done by analyzing the logs.

zhangyanjiaoesp, 2025-09-16:
> Could you confirm if this issue is caused by RSSI fluctuations, possibly due to using the same channel, which in turn triggers parent changes, Should I change RSSI Threshold, to handle this?

@mightChamp I think you can change channel and try again. change RSSI threshold also can help.

zhangyanjiaoesp, 2025-09-19:
> ```
> 	/* Disable mesh PS function */
> 		ESP_ERROR_CHECK(esp_mesh_disable_ps());
> 		ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(10));
> ```
> 
> can esp_mesh_set_ap_assoc_expire() play role in connect disconnect of nodes, as each device transmit a data frame each 30 min to root node.

Yes, it will, please refer to the comments of the API.

[image]

zhangyanjiaoesp, 2025-09-19:
By the way, are you using LR only mode or BGNLR mode? Suggest you use the BGNLR mode, this way, when the communication environment is good, you can transmit at higher rates, and when the environment is poor, the rate will fall back to LR for data transmission.

mightChamp, 2025-09-19:
@zhangyanjiaoesp Thanks for support, it helps lot. By Switching channel efficiency increased to nearly 90%. Need some more work to do for 100%.