espressif/esp-idf #17595: Cannot perform OTA update from app that built on ESP-IDF >=v5.4.2 and v5.5.1

dmitrij9992905, 2025-09-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

>=v5.4.2, v5.5.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

If the project was originally built on IDF <=v5.4.1 (app+bootloader), then app built on v5.4.2 or newer should be updated as well

### What is the actual behavior?

On OTA update request:
IDF v5.4.2:
```
Couldn't allocate memory to upgrade data buffer
```
IDF v5.5.1:
```
...
I (13:02:26.574) esp_https_ota: Starting OTA... 
E (13:02:26.575) esp_https_ota: Given staging partition or another suitable Passive OTA partition could not be found 
E (13:02:26.583) OTA: ESP HTTPS OTA Begin failed
...
```

And OTA update ends with error

### Steps to reproduce.

1. Originally build the project (app+bootloader) on ESP-IDF v5.4.1 or older
2. Flash entire project into ESP32S3
3. Rebuild the project using ESP-IDF v5.4.2 or newer
4. Try to perform OTA update [it performs successfully]
5. Build another version of project using ESP-IDF v5.4.2 or newer
6. Try to perform OTA update [failure goes here] 

### Debug Logs.

```plain

```

### More Information.

[CMakeLists.txt](https://github.com/user-attachments/files/22383431/CMakeLists.txt)
[ota_management.cpp](https://github.com/user-attachments/files/22383432/ota_management.cpp)
[partition_table_8mb.csv](https://github.com/user-attachments/files/22383444/partition_table_8mb.csv)

The devices with the partition table are already in production, as at least the OTA was OK

dmitrij9992905, 2025-09-18:
For ota_management.cpp, it worked

I'll also try with some device with old bootloaders that we have as testing devices

dmitrij9992905, 2025-09-19:
For HTTPS, that's the issue:
```
I (10:42:04.204) OTA: Creating OTA update task
W (10:42:04.204) OTA: No OTA deferring signs, continuing...
I (10:42:04.217) OTA: OTA client initiated and ready to download a new img
...
I (10:42:04.580) esp-x509-crt-bundle: Certificate validated
...
I (10:42:05.043) esp-x509-crt-bundle: Certificate validated
...
E (10:42:05.394) esp-tls-mbedtls: read error :-0x7100
E (10:42:05.395) transport_base: esp_tls_conn_read error, errno=Success
...
E (10:42:05.401) esp_https_ota: Failed to establish HTTP connection
E (10:42:05.416) OTA: ESP HTTPS OTA Begin failed
W (10:42:05.417) OTA: The update will be retried in 73 sec
```

mahavirj, 2025-09-29:
@dmitrij9992905 

1. Please check OTA troubleshooting pointers [here](https://github.com/espressif/esp-idf/tree/master/examples/system/ota#troubleshooting)
2. Suggest checking ESP-IDF supplied OTA example once
3. If you still run into any issues then please share your `sdkconfig`, example project and mbedtls debug enabled console log

dmitrij9992905, 2025-09-29:
Hello everyone!

Sorry for long reply.
For now, OTA works normally over HTTP, and HTTPS not tested properly from S3 storage.
If I face with HTTPS problem again, I'll reopen the issue.