espressif/esp-idf #17600: Use WIFI_PROTOCOL_11AX protocol in WiFi-Mesh with ESP32-C6

mightChamp, 2025-09-18:
### Is your feature request related to a problem?

I need Wi-Fi 6 Support in Wi-Fi-Mesh for robustness, and better connectivity.
So I set protocol using
```
ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA));
ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11AX));
ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11AX));
````

But 
```
ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11AX));
```

This generates error in run time, as WIFI_PROTOCOL_11AX not supported in AP Mode,

So I would like to use WIFI_PROTOCOL_11AX in Wi-Fi Mesh Network.

### Describe the solution you'd like.

Provide support for ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11AX));

### Describe alternatives you've considered.

Currently using LR Mode

zhangyanjiaoesp, 2025-09-19:
@mightChamp 
you need to call `esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_11AX);`
esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11AX) will return error.

mightChamp, 2025-09-19:
@zhangyanjiaoesp  

Are you recommending the use of the following configuration to address network issues, automate protocol management, and improve overall efficiency?

```
esp_wifi_set_protocol(WIFI_IF_AP,  WIFI_PROTOCOL_LR | WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_11AX);
esp_wifi_set_protocol(WIFI_IF_STA,  WIFI_PROTOCOL_LR | WIFI_PROTOCOL_11B | WIFI_PROTOCOL_11G | WIFI_PROTOCOL_11N | WIFI_PROTOCOL_11AX);
```

zhangyanjiaoesp, 2025-09-22:
> [@zhangyanjiaoesp](https://github.com/zhangyanjiaoesp)
>
> Are you recommending the use of the following configuration to address network issues, automate protocol management, and improve overall [...]

yes

mightChamp, 2025-09-22:
@zhangyanjiaoesp Thanks for support

closed 2025-09-22 by mightChamp