espressif/esp-idf #17606: ESP32-S3 + RTL8201

janav1, 2025-09-19:
Hi, I am looking for help from more experienced developers.
I'm trying to design a board with ESP32-S3-MINI-1U and RTL8201F Ethernet PHY. I was inspired by LilyGo T-ETH_LITE:

[schematic](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-Series/blob/master/schematic/T-ETH-Lite-ESP32.pdf)

but I'm not sure about connecting the clock line between ESP and RTL.
The RTL8201F provides a reference clock output on the TXC pin, but I don't know where to connect this pin on the ESP side (input to the RMII). I will use the RMII interface.
I didn't find any mention in the ESP32-S3 documentation about which pin the clock can be connected to. Can any pin be used in combination with the GPIO matrix feature?

The second question concerns the inverter used on the LilyGo board. Is this also necessary for the ESP32-S3? I understood that this was due to maintain the necessary timing.
Thank you all for the advice.

dmitrij9992905, 2025-09-19:
ESP32-S3 does not support (R)MII interface to connect external Ethernet PHYs like RT8201F.
It supported by ESP32 and ESP32P4.

As an option, you can use SPI-Ethernet adapter

janav1, 2025-09-21:
that's a pity :-(
When using ESP32, the only option to connect the RMII clock signal is GPIO0, right?
What alternative Ethernet chip would you recommend? I'm thinking about the Wiznet W6300 and QSPI mode.

dmitrij9992905, 2025-09-21:
As for SPI-Ethernet, I use W5500 in my designs. As for througput, I managed to get ~8 Mbit/s. I think it's sufficient for most applications.

As for ESP32&RMII, as far as I know, it's possible to use IO0 both as input for RMII clock and output for reference RMII clock for PHY.

closed 2025-10-06 by kostaond