espressif/esp-idf #17629: [v5.1.4] Yamaha CP88  - EP0 control transfer fails with USB_TRANSFER_STATUS_STALL for Config descriptor

chipweinberger, 2025-09-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom

### Power Supply used.

USB

### What is the expected behavior?

read the configuration descriptor

### What is the actual behavior?

the transfer fails with USB_TRANSFER_STATUS_STALL. 

```
[0;31mE (959019) : e1 : [pd usbh] USB TRANSFER FAILED - status: Stall
e2 : [pd usbh [device]] pd_usbh_service_submitTransferAndWait() failed.
e3 : [pd midi usb] pd_usbh_device_getConfigDescriptor() failed.
```

### Steps to reproduce.

1. plug in Yamaha CP88
2. try to read config descriptor

### Debug Logs.

```plain
[0;31mE (959019) : e1 : [pd usbh] USB TRANSFER FAILED - status: Stall
e2 : [pd usbh [device]] pd_usbh_service_submitTransferAndWait() failed.
e3 : [pd midi usb] pd_usbh_device_getConfigDescriptor() failed.
 [0m
```

roma-jam, 2025-09-22:
Hi @chipweinberger, 

thanks for the report. 

Could you please share the raw data request you are using to get the configuration descriptor?
Is it `80 06 01 02 00 00 XX 00`? 

I can see that device has `bNumConfigurations: 2` but that doesn't mean that the their indexes are 1..N or sequential.

To get the correct value, we need to get the `bConfigurationValue` from the descriptor. To do so we need to read descriptors for each index `i = 0..bNumConfigurations-1` and then to use the values we read. 

Also, if that is possible, you could enable the VERBOSE log level and the usb debug will have the raw requests data. This could be used as an option to verify which request is being sent. 

Thanks.

chipweinberger, 2025-09-22:
> I can see that device has bNumConfigurations: 2 but that doesn't mean that the their indexes are 1..N or sequential.

Ah, good point.

That said my code still does the right thing despite using the wrong name in the logs.

I'm trying to read the second configuration descriptor (by index), which should exist because `bNumConfigurations=2`, but I get a stall.

**For reference:**
```
// submits control transfer
xesp_usb_config_descriptor_t* pd_usbh_device_getConfigDescriptor(pd_usbh_device_t usbhDevice, uint8_t bConfigurationValue, pd_error_t *error)
{
    pd_err_unset(error);

    // allocations
    xesp_usb_config_descriptor_t* parsed = NULL;
    pd_usb_transfer_t* transfer = pd_usbh_service_takeTransfer();
    {
        ESP_LOGI(TAG, "getConfigDescriptor(bConfigurationValue: %" PRIu8 ")", bConfigurationValue);

        uint8_t* configDesc = NULL;
        uint32_t length = 0;

        // get cached confguration
        const usb_config_desc_t* cachedConfigDesc;
        sTry(esp_err_t err =,usb_host_get_active_config_descriptor,(usbhDevice.deviceHandleB, &cachedConfigDesc));

        // use cache?
        if (bConfigurationValue == cachedConfigDesc->bConfigurationValue) {

            ESP_LOGI(TAG, "using cached configuration!");
            ESP_LOGI(TAG, "wTotalLength %" PRIu16, cachedConfigDesc->wTotalLength);

            // data
            configDesc = (uint8_t*) cachedConfigDesc;

            // length
            length = cachedConfigDesc->wTotalLength;

        } else {

            // Step 1: Get short config descriptor (just the header)
            // Usually the USB configuration descriptor length is at least 8 bytes.
            // This gives us wTotalLength which we use for the second request.
            const size_t SHORT_DESC_REQ_LEN = 8;

            // Initialize the GET_CONFIG_DESC setup packet for the SHORT read
            USB_SETUP_PACKET_INIT_GET_CONFIG_DESC((usb_setup_packet_t *)transfer->val->data_buffer, 
                                                bConfigurationValue,
                                                SHORT_DESC_REQ_LEN);

            transfer->val->num_bytes = sizeof(usb_setup_packet_t) + SHORT_DESC_REQ_LEN;
            transfer->val->flags = 0;
            transfer->val->device_handle = usbhDevice.deviceHandleB;
            transfer->val->bEndpointAddress = 0;
            transfer->val->status = (usb_transfer_status_t) 0;
            transfer->val->timeout_ms = 500;
            transfer->val->context = 0;

            eTry(, pd_usbh_service_submitTransferAndWait, (transfer, error));

            // Check the actual number of bytes returned
            if (transfer->val->actual_num_bytes < sizeof(usb_setup_packet_t) + SHORT_DESC_REQ_LEN) {
                eFail("[SHORT_DESC_REQ] insufficient bytes: %i", transfer->val->actual_num_bytes);
                goto end;
            }

            // Extract the short descriptor data
            uint8_t* shortDesc = transfer->val->data_buffer + sizeof(usb_setup_packet_t);

            // The first 9 bytes represent a usb_config_desc_t structure:
            //  offset 2-3: wTotalLength (LE)
            uint16_t wTotalLength = shortDesc[2] | ((uint16_t)shortDesc[3] << 8);

            ESP_LOGI(TAG, "wTotalLength: %u", wTotalLength);

            // Check if we can handle wTotalLength with our buffer (if necessary)
            if (wTotalLength > CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE) {
                eFail("wTotalLength > CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE");
                goto end;
            }

            // Step 2: Get the full configuration descriptor
            USB_SETUP_PACKET_INIT_GET_CONFIG_DESC((usb_setup_packet_t *)transfer->val->data_buffer, 
                                                bConfigurationValue,
                                                wTotalLength);

            transfer->val->num_bytes = sizeof(usb_setup_packet_t) + wTotalLength;
            transfer->val->flags = 0;
            transfer->val->device_handle = usbhDevice.deviceHandleB;
            transfer->val->bEndpointAddress = 0;
            transfer->val->status = (usb_transfer_status_t) 0;
            transfer->val->timeout_ms = 500;
            transfer->val->context = 0;

            eTry(,pd_usbh_service_submitTransferAndWait, (transfer, error));

            // data
            configDesc = transfer->val->data_buffer + sizeof(usb_setup_packet_t);

            // length
            length = transfer->val->actual_num_bytes - sizeof(usb_setup_packet_t);

            // check length
            if (length < wTotalLength) {
                // Device returned less than expected. According to USB spec, this is not allowed.
                ESP_LOGE(TAG, "fewer bytes than expected. wTotalLength=%" PRIu16 ", got=%" PRIu32, wTotalLength, length);
                eFail("Incomplete full descriptor");
                goto end;
            }
        }

        // log
        printHexLines("config descriptor", configDesc, length, 32);

        // Step 3: Parse the full config descriptor
        parsed = xesp_usbh_parse_config(configDesc, length, error);
        if (!parsed) {
            eFail("Failed to parse config descriptor");
            goto end;
        }

        pd_err_success(error);
    }

end:
    pd_usbh_service_giveTransfer(transfer);
    eOnFail(return NULL);
    return parsed;
}
```

closed 2025-09-25 by roma-jam