espressif/esp-idf #17646: Transition from Station to AP causing issues

KaeLL, 2025-09-24:
I'm having an issue switching the Wi-Fi from Station mode to AP mode, in that the AP_STOP event is apparently triggered externally (not by me or any esp_wifi_stop from my code) after the AP has started, but then it's started a second time.
The sequence of events is STA_STOP -> AP_START -> AP_STOP -> AP_START, with the last two being undesirable, and happening just right after the AP_START event.

Here's the example code, along with the full log of a test run
[test_wifi.zip](https://github.com/user-attachments/files/22504759/test_wifi.zip)

AkshatAgrawl, 2025-09-25:
Hi @KaeLL , I read your code. Seems you should deinitialize STA and stop Wi-Fi properly before starting AP. The redundant AP-STOP → AP-START happens because Wi-Fi isn’t fully stopped. Please follow this clean sequence:
```
ESP_ERROR_CHECK(esp_wifi_stop());
ar_wifi_sta_deinit();

ar_wifi_ap_init();
ar_wifi_ap_start();
```

KaeLL, 2025-09-25:
That worked. Thanks @AkshatAgrawl

closed 2025-09-25 by KaeLL