espressif/esp-idf #17650: SPI Slave API makes no sense

beelsebob, 2025-09-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.5

### Espressif SoC revision.

ESP32-S2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I'm trying to figure out how the ESP IDF SPI Slave API works, and the documentation makes absolutely no sense.  I'm expecting that I would register as a slave on a certain SPI host, and from then on receive callbacks whenever the master sends me data, asking "what data do you want to send back?"  I use `spi_slave_initialize`, but the callbacks provided appear to never be called.  How on earth is this meant to work?

### What is the actual behavior?

You use `spi_slave_initialize`, and then no callbacks occur, despite master sending data to you.

### Steps to reproduce.

Call spi_slave_initialize, and wait for callbacks.  Attach another device sending spi data to it.

### Debug Logs.

```plain

```

ginkgm, 2025-09-27:
Hi @beelsebob ,

By design, the driver itself is not event driven, but synchronized to reduce the complexity of callbacks for the users. This means  that it's not necessary to use callbacks in most cases, and it's also not recommended/not allowed to queue new transactions in the callbacks.

Please refer to the expected usage in the example: https://github.com/espressif/esp-idf/blob/master/examples/peripherals/spi_slave/receiver/main/app_main.c

You are supposed to prepare a transaction in correspond to master's following transaction, and you can respond to master's transaction by checking the result of https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/spi_slave.html#_CPPv426spi_slave_get_trans_result17spi_host_device_tPP23spi_slave_transaction_t10TickType_t . When the SW has not prepared the transaction, the driver is not listening on the bus. 

When the callback was introduced, it was designed to provide an OOB (out of band) signal to the master that "I have prepared the data and you can start transaction". Not to implement an event driven scenario. You can see in the code that it's tightly coupled to the transaction provided by the user - when there is no transaction prepared, the callbacks are not supposed to be called : https://github.com/espressif/esp-idf/blob/master/components/esp_driver_spi/src/gpspi/spi_slave.c#L728

If you do want the hardware to keep listening on the bus and always reply something on transaction, you can try another driver: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/spi_slave_hd.html#_CPPv430spi_slave_hd_callback_config_t . There are more events for you. But please note that, there are more restrictions for ISR callbacks.

wanckl, 2025-09-29:
@beelsebob Hi, As you expect the slave's behavior: master_trans -> trans_finish -> slave_interrupt -> slave_read_data,    

yeah it is a classic mode, but it also means the hardware fifo must hold the data between trans_finish and slave_read_data, this suitable for some small data protocols like i2c, twai.

1. spi is more tend to mess data transfer that overflow the hardware fifo, so provide tx/rx buffer before transaction is expected.

2. spi is a full duplex transaction, even you just want to receive something after transaction, what to send still need to be defined even if a NULL