espressif/esp-idf #17654: esp_blockdev suggestions

BitsForPeople, 2025-09-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I appreciate the new `esp_blockdev` component.
I'd like to suggest a few changes before it's "closed" for modification:

1. Allow for future custom IOCTL commands, i.e. the `ioctl` function should accept the command as an `int` instead of `esp_blockdev_ioctl_cmd_t`
2.  A block device instance should provide a way for the driver to attach driver-specific "meta-data" to the instance, like a drive number, a partition name, or something else. A `void* dev_ctx;` field in `struct esp_blockdev_t` should be sufficient and would simplify driver implementation.
3. `ESP_BLOCKDEV_CMD_TRIM` - TRIM is an operation on a _filesystem_, i.e. it's nothing a _block device_ can or should be requested to do
4. The block device's functions should be factored out into their own struct, which should be referenced from `struct esp_blockdev_t` as a const pointer, e.g. as `const esp_block_dev_ops_t* ops;`. The functions are an immutable property of the respective device driver and known at compile-time, i.e. they should be able to be held in a single global, const, statically-allocated structure per driver, not per block device instance.

igrr, 2025-09-25:
Thanks for those suggestions @BitsForPeople!

pacucha42, 2025-10-07:
The update is available in IDF master branch, closing the tracker.

Feel free to reopen or open a new one.

closed 2025-10-07 by pacucha42